phanrahan/magma #87: Broken register simulation

cdonovick, 2017-08-29:
```
OPCODE_W=6
BIT_W=4

OPCODE_T=m.In(m.UInt(OPCODE_W))
I_T = m.In(m.UInt(BIT_W))
O_T = m.Out(m.UInt(BIT_W))

OReg = DefineRegister(BIT_W, T=m.UInt)

from functools import reduce

def one_hot_mux(cv):
    outputs = []
    for cond, inp in cv:
        outputs.append(inp & [cond for _ in range(len(inp))])
    return list(reduce(lambda x, y: x | y, outputs))

_AND = m.uint(0, OPCODE_W)
_OR  = m.uint(1, OPCODE_W)
_XOR = m.uint(2, OPCODE_W)
_ADD = m.uint(3, OPCODE_W)
_SUB = m.uint(4, OPCODE_W)
_NEG = m.uint(5, OPCODE_W)
_INV = m.uint(6, OPCODE_W)

op_funcs = {
    _AND : lambda a, b : a&b,
    _OR  : lambda a, b : a|b, 
    _XOR : lambda a, b : a^b,
    _ADD : lambda a, b : a+b,
    _SUB : lambda a, b : a-b,
    _INV : lambda a, b : ~a,
}

class ALU(m.Circuit):
    name = "ALU"
    IO = ["opcode", OPCODE_T, "A", I_T, "B", I_T, "R", O_T] + m.ClockInterface()
    
    @classmethod
    def definition(io):
        cv = [(io.opcode == o, op_funcs[o](io.A, io.B)) for o in op_funcs]
        o_reg = OReg()
        m.wire(one_hot_mux(cv), getattr(o_reg, "in"))
        m.wire(o_reg.out, io.R)
        m.wireclock(io, o_reg)
```
```
from magma.python_simulator import PythonSimulator
from magma.scope import Scope
from magma.bit_vector import BitVector

simulator = PythonSimulator(ALU)
scope = Scope()

A_vals = range(4)
B_vals = range(4)

Op_in = []
A_in = []
B_in = []
R_out = []
E_out = []

simulator.set_value(ALU.opcode, scope, BitVector(1, OPCODE_W))

for a in A_vals:
    simulator.set_value(ALU.A, scope, BitVector(a, BIT_W))
    for b in B_vals:
        simulator.set_value(ALU.B, scope, BitVector(b, BIT_W))
        for _ in range(2):
            simulator.evaluate()
            r = BitVector(simulator.get_value(ALU.R, scope), BIT_W).as_int()                
            A_in.append(a)
            B_in.append(b)

            R_out.append(r)

            print('{}({}, {})={}'.format('or', a,b,r))
```

```
or(0, 0)=0
or(0, 0)=0
or(0, 1)=0
or(0, 1)=0
or(0, 2)=0
or(0, 2)=0
or(0, 3)=0
or(0, 3)=0
or(1, 0)=0
or(1, 0)=0
or(1, 1)=0
or(1, 1)=0
or(1, 2)=0
or(1, 2)=0
or(1, 3)=0
or(1, 3)=0
or(2, 0)=0
or(2, 0)=0
or(2, 1)=0
or(2, 1)=0
or(2, 2)=0
or(2, 2)=0
or(2, 3)=0
or(2, 3)=0
or(3, 0)=0
or(3, 0)=0
or(3, 1)=0
or(3, 1)=0
or(3, 2)=0
or(3, 2)=0
or(3, 3)=0
or(3, 3)=0
```

cdonovick, 2017-08-29:
```
import magma as m
from magma.primitives import DefineRegister
from mantle.lattice.mantle40 import MUX

USE_REGISTER=True
```

```
OPCODE_W=3
BIT_W=16

OPCODE_T=m.In(m.UInt(OPCODE_W))
I_T = m.In(m.UInt(BIT_W))
O_T = m.Out(m.UInt(BIT_W))

OReg = DefineRegister(BIT_W, T=m.UInt)

_NOP = 0
_AND = 1
_OR  = 2
_XOR = 3
_ADD = 4
_SUB = 5
_INV = 6
_MUL = 7

_op_funcs = {
    _NOP : lambda a, b : m.uint(0, BIT_W),
    _AND : lambda a, b : a&b,
    _OR  : lambda a, b : a|b, 
    _XOR : lambda a, b : a^b,
    _ADD : lambda a, b : a+b,
    _SUB : lambda a, b : a-b,
    _INV : lambda a, b : ~a,
    _MUL : lambda a, b : a*b,
}

_op_names = {
    _NOP : "nop",
    _AND : "and",
    _OR  : "or",
    _XOR : "xor",
    _ADD : "add",
    _SUB : "sub",
    _INV : "inv",
    _MUL : "mul"
}

class ALU(m.Circuit):
    name = "ALU"
    IO = ["opcode", OPCODE_T, "A", I_T, "B", I_T, "R", O_T] + m.ClockInterface()
    
    @classmethod
    def definition(io):

        mux_inputs = [_op_funcs[i](io.A, io.B) for i in range(len(_op_funcs))] + [io.opcode]
        op_mux = MUX.Mux(height=len(_op_funcs), width=BIT_W)
        if USE_REGISTER:
            o_reg = OReg()
            m.wire(op_mux(*mux_inputs), getattr(o_reg, "in"))
            m.wire(o_reg.out, io.R)
            m.wireclock(io, o_reg)
        else:
            m.wire(op_mux(*mux_inputs), io.R)
```

```
from magma.python_simulator import PythonSimulator
from magma.scope import Scope
from magma.bit_vector import BitVector

simulator = PythonSimulator(ALU)
scope = Scope()

A_vals = range(4)
B_vals = range(4)

for op in _op_funcs:
    simulator.set_value(ALU.opcode, scope, BitVector(op, OPCODE_W))
    for a in A_vals:
        simulator.set_value(ALU.A, scope, BitVector(a, BIT_W))
        for b in B_vals:
            simulator.set_value(ALU.B, scope, BitVector(b, BIT_W))
            for _ in range(2):
                if USE_REGISTER:
                    simulator.step()
                simulator.evaluate()
                r = BitVector(simulator.get_value(ALU.R, scope), BIT_W).as_int()

                print('{}({}, {})={}'.format(_op_names[op], a,b,r))
```

leonardt, 2017-08-29:
Resolved in person (step semantics are confusing).

closed 2017-08-29 by leonardt