phanrahan/magma #96: Error when wiring sequence to Type or vice versa

Syn-X-Lin, 2017-09-03:
After recent commits, the following snippet will return an error:
```python
wire(int2seq(1, n), adder.I1)
```
The error info is
```
File "/home/tmp/magma_hackathon/magma/magma/debug.py", line 24, in wire
    return fn(i, o, debug_info)
File "/home/tmp/magma_hackathon/magma/magma/wire.py", line 31, in wire
    if not isinstance(o, IntegerTypes) and o.isinput():
AttributeError: 'list' object has no attribute 'isinput'
```
It seems that `wire.py` cannot handle this situation after commenting out
```python
if isinstance(i, Sequence):
    if isinstance(o, ArrayType) and len(i) == len(o):
```

phanrahan, 2017-09-03:
Thanks for the bug report. int2seq is depreciated. That should be bits(i,n) or uint(i,n). I will fix it.

Syn-X-Lin, 2017-09-03:
Is there any way to perform the following operation effectively after removing wire operation for list ?
'''python
# io.I is an input of  4 bits.
# lut is an instance of SB_LUT4
wire(io.I, [lut.I0, lut.I1, lut.I2, lut.I3])
'''

leonardt, 2017-09-04:
```
wire(io.I, bits([lut.I0, lut.I1, lut.I2, lut.I3])
```

closed 2017-09-05 by phanrahan