phanrahan/magma #97: Error when using multiple mantle comparison modules

Syn-X-Lin, 2017-09-03:
It will return error when using multiple mantle comparison modules in one script.
The error message is
```
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/compare.py", line 30, in EQ
    return Cascade(n, 2, A0&((A1&A2)|(~A1&~A2)), 1, **kwargs)
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/cascade.py", line 48, in Cascade
    return DefineCascade(n, k, expr, cin, forkargs)(**kwargs)
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/cascade.py", line 27, in DefineCascade
    class _Cascade(Circuit):
  File "/home/tmp/magma_hackathon/magma/magma/circuit.py", line 358, in __new__
    warn('Warning: Circuit {} already defined'.format(name))
  File "/home/tmp/magma_hackathon/magma/magma/error.py", line 23, in warn
    traceback.print_stack()
Warning: Circuit Invert4 already defined
```

```
File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/compare.py", line 61, in ULE
    sub = Sub(n, **kwargs)
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/arith.py", line 17, in Sub
    invert = Invert(n)
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/logic.py", line 364, in Invert
    return DefineInvert(n)(**kwargs)
  File "/home/tmp/magma_hackathon/mantle/mantle/lattice/mantle40/logic.py", line 348, in DefineInvert
    class _Invert(Circuit):
  File "/home/tmp/magma_hackathon/magma/magma/circuit.py", line 358, in __new__
    warn('Warning: Circuit {} already defined'.format(name))
  File "/home/tmp/magma_hackathon/magma/magma/error.py", line 23, in warn
    traceback.print_stack()
Warning: Circuit Mux2x16 already defined
```

phanrahan, 2017-09-04:
I added that warning. I see know that that prints a stack trace and exits. It should not do that. May want to take out that warning.

phanrahan, 2017-09-05:
Removed the warning.

closed 2017-09-05 by phanrahan