phanrahan/magma #99: Shift register example from tests contains instance that coreir does not recognize

dillonhuff, 2017-09-06:
I've been trying to generate coreir json from magma circuits to get larger examples for the simulator.

When I run the following example (ShiftRegister):

```python
from magma import *
from magma.primitives import DefineRegister

from magma.verilator.verilator import compile as compileverilator

from magma.scope import Scope
from magma.bit_vector import BitVector
from magma.bitutils import int2seq, seq2int

from magma.transforms import flatten, setup_clocks

N = 4
Register4 = DefineRegister(4)
T = Bits(N)

class ShiftRegister(Circuit):
    name = "ShiftRegister"
    IO = ["I", In(T), "O", Out(T), "CLK", In(Clock)]
    @classmethod
    def definition(io):
        regs = [Register4() for _ in range(N)]
        wireclock(io, regs)
        wire(io.I, getattr(regs[0], "in"))
        fold(regs, foldargs={"in":"out"})
        wire(regs[-1].out, io.O)

compile("shift_register_verilog", ShiftRegister)
compile("shift_register_coreir", ShiftRegister, 'coreir')

```

I get the following output:

```bash
(magma_venv) bash-3.2$ python shift_reg.py
python shift_reg.py
('compiling', 'ShiftRegister')
ERROR: Could not find Instance in library!
  Instance: reg_P
  Namespace: coreir

I AM DYING!
```

Am I doing something wrong here? The magma to verilog compilation works for this circuit, but coreir does not.

leonardt, 2017-09-06:
Looks like a bug in the register primitive definition, I will investigate

leonardt, 2017-09-06:
Should be fixed by 3c52fe84dc2e1bcda9d7878a3caaa3f88705c6c8, can you try it out?

dillonhuff, 2017-09-06:
Just checked. It works now.

Thanks!

leonardt, 2017-09-06:
The issue was that the verilog_name of the coreir module didn't match the coreir_prims name (because in Verilog the permutations of reg types is based on the names, whereas in coreir it's a generator arguments). Fixed by adding an explicit coreir_name instead of just trying to use the verilog_name

closed 2017-09-06 by leonardt