phanrahan/magma #101: COUT skipped in interface.outputs()

leonardt, 2017-09-09:
Currently `CircuitType.interface.outputs()` skips outputs named `COUT` (see https://github.com/phanrahan/magma/blob/master/magma/interface.py#L105-L107). I'm guessing this is useful for some higher order composition of `AddC` modules, but it prevents the following code from working:
```
    T = UInt(4)
    class TestCircuit(Circuit):
        name = "test_addc"
        IO = ["a", In(T), "b", In(T), "c", Out(T), "d", Out(Bit)]
        @classmethod
        def definition(circuit):
            out, carry = AddC(2, 4)(circuit.a, circuit.b)
            wire(out, circuit.c)
            wire(carry, circuit.d)
```

@phanrahan can you describe an example where skipping `COUT` arguments is required? Any ideas on a workaround that would allow the above code to work? (Return `COUT` as one of the outputs to a call function for wiring).

phanrahan, 2017-09-09:
Yes, that is a bit kludgy. It had to do with higher order operators and special handling
of COUT in counters.

But I propose to remove it. I will try to do that today. It will probably break a bunch of things!

Btw, when looking at the code in lattice/mantle40/arith.py I noticed the way carrys were
being handled was not the way I think it should be. I propose the following interfaces

```
def Add(n, cin=False, cout=False, **kwargs):
    return Adders(n, cin, cout, **kwargs)

def AddC(n, **kwargs):
    return Add(n, True, True, **kwargs)

def Sub(n, cin=False, cout=False, **kwargs):
    invert = Invert(n)
    adder =  Adders(n, cin, cout, **kwargs)
    wire(invert.O, adder.I1)
    args = ["I0", adder.I0, "I1", invert.I, "O", adder.O]
    if cin:
        args += ['CIN', adder.CIN]
    if cout:
        args += ['COUT', adder.COUT]
    return AnonymousCircuit(args)

def SubC(n, **kwargs):
    return Sub(n, True, True, **kwargs)
```

This means Adders is no longer exposed in mantle40, since you can do everything in Adders.
We could probably even get rid of Adders and just use Add in mantle.common.

phanrahan, 2017-09-09:
I changed interface.py so that CIN and COUT are handled as normal inputs and outputs.

That required changes to mantle. See the mantle.common and mantle.lattice.mantle40.

In the process, I depreciated Adders, and changed the interface to Add and Sub.

rdaly525, 2017-09-09:
What are your thoughts on changing the coreir add/sub to have an optional cout option? How are you currently mapping any adders that utilize cin/cout to coreir primitives?

phanrahan, 2017-09-11:
On further thought, I propose just two primitives:

Add(n, cin, cout)
Sub(n, cin, cout)

cin and cout are bools that indicate whether to include a cin or cout port.

leonardt, 2017-09-11:
@rdaly525 cout/cin in the primitive definition would be useful. For COUT, I promote the input widths by 1 by concatting a zero to the front. With this, the msb bit in the output to be the carry. For CIN, I concat it with width-1 zeros and insert another add. Not sure if there's better mapping scheme (particuarly for CIN), but it seems like support in the primitive directly would make this simpler.

leonardt, 2017-09-11:
Just realized I should be sign extending instead of zero extending to implement the above carry logic.  It seems that the choice of carry logic implementation is platform dependent so I will vote to have the coreir primitive have an explicit notion of CIN and COUT

leonardt, 2017-09-15:
Closing this because it's not longer an issue

closed 2017-09-15 by leonardt