phanrahan/magma #103: wireclock doesn't wire Clock types

leonardt, 2017-09-15:
The current definition doesn't wire `Clock` types, only Reset and Enable. I have to use `wiredefaultclock` for automatically wiring `Clock`s, is this by design?

phanrahan, 2017-09-15:
Clocks are wired by default in the compiler. We should review the way this is being done.

leonardt, 2017-09-15:
Does that conflict with having an explicit `wireclock` call wire up the Clock port (that is, if it's wired up by the time it hits default wiring logic in the compiler, does it cause an issue)? 

It seems like if we want to leave the default clock wiring up to the compiler we might want to change the name of `wireclock` as right now the behavior doesn't match the name of the function. Unfortunately I can't think of a better name.

leonardt, 2020-08-07:
Clock wiring is done internally and this issue has been resolved.

closed 2020-08-07 by leonardt