phanrahan/magma #120: magma -> coreir should not be split into a bunch of 1 bit registers/wires

rdaly525, 2017-10-25:
For example a magma->coreir compiled counter looks like the following:

```
{"top":"global.Counter4_flattened",
"namespaces":{
  "global":{
    "modules":{
      "Counter4_flattened":{
        "type":["Record",{
          "O":["Array",4,"Bit"],
          "COUT":"Bit",
          "CLK":["Named","coreir.clkIn"]
        }],
        "instances":{
          "bit_const_GND":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",false]}
          },
          "bit_const_VCC":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",true]}
          },
          "inst0":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",5]}
          },
          "inst1":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",false], "has_rst":["Bool",false], "width":["Int",1]}
          },
          "inst2":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",false], "has_rst":["Bool",false], "width":["Int",1]}
          },
          "inst3":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",false], "has_rst":["Bool",false], "width":["Int",1]}
          },
          "inst4":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",false], "has_rst":["Bool",false], "width":["Int",1]}
          }
        },
        "connections":[
          ["bit_const_GND.out","inst0.in0.0"],
          ["bit_const_GND.out","inst0.in1.0"],
          ["bit_const_GND.out","inst0.in1.2"],
          ["bit_const_GND.out","inst0.in1.3"],
          ["bit_const_GND.out","inst0.in1.4"],
          ["bit_const_VCC.out","inst0.in1.1"],
          ["inst1.clk","self.CLK"],
          ["inst2.clk","self.CLK"],
          ["inst3.clk","self.CLK"],
          ["inst4.clk","self.CLK"],
          ["self.COUT","inst0.out.4"],
          ["inst0.in0.1","inst1.out.0"],
          ["inst0.in0.2","inst2.out.0"],
          ["inst0.in0.3","inst3.out.0"],
          ["inst0.in0.4","inst4.out.0"],
          ["inst0.out.0","inst1.in.0"],
          ["inst0.out.1","inst2.in.0"],
          ["inst0.out.2","inst3.in.0"],
          ["inst0.out.3","inst4.in.0"],
          ["inst1.out.0","self.O.0"],
          ["inst2.out.0","self.O.1"],
          ["inst3.out.0","self.O.2"],
          ["inst4.out.0","self.O.3"]
        ]
      }
    }
  }
}
}
```

I am not sure why 1-bit registers are being instantiated rather than a 5-bit register. It makes the produced code difficult to read and likely slow to simulate. 
What is the reason the code looks like this?
Perhaps the issue is that the concat operation flattens arrays to bits? If so, can we add a concat operator in magma?

leonardt, 2017-10-25:
This is a mantle issue, but what's happening is that `mantle.common` has a `Register` definition that defines a generic `N`-bit register in terms of an `FF` primitive. So, `mantle.coreir` implements `FF` and then `mantle.common.Register` defines that circuit using `FF`.

This looks like a case where we would want the `mantle.coreir` implementation to be able to provide a `specialized` implementation of `Register`

leonardt, 2017-10-25:
Moved to https://github.com/phanrahan/mantle/issues/40

closed 2017-10-25 by leonardt