phanrahan/magma #133: Trouble Converting Magma To CoreIR and Running on Icestick

David-Durst, 2018-01-14:
What is the series of tools I need to go through CoreIR instead of Verilog when putting Magma code on the Icestick?

My current hypothesis is Magma -> coreIR -> yosys -> arachne-pnr -> icepack -> iceprog. However, I don't understand the Magma -> coreIR step, nor am I confident that I know what to do with the coreIR JSON. 

# Problem 1: How to Convert Magma to CoreIR

I tried in https://github.com/phanrahan/magma/blob/master/notebooks/icestick/counter.ipynb replacing In[5] with m.compile("build/ice_counter", main, output='coreir') and m.compile("build/ice_counter", main, output='coreir', include_coreir=True) to output coreIR instead of verilog. These seemed like the correct options based on the help output of the magma binary. When I ran this, my IPython Notebook kernel crashed and I got the following error:

ERROR: Param type mismatch for: LUT_INIT (38550 vs BitVector<16>)

0   libcoreir.dylib                     0x000000011057f7a4 _ZN6CoreIR20checkValuesAreParamsENSt3__13mapINS0_12basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEPNS_5ValueENS0_4lessIS7_EENS5_INS0_4pairIKS7_S9_EEEEEENS1_IS7_PNS_9ValueTypeESB_NS5_INSC_ISD_SI_EEEEEE + 14484
1   libcoreir.dylib                     0x00000001107b1b40 _ZN6CoreIR8InstanceC2EPNS_9ModuleDefENSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEEPNS_6ModuleENS3_3mapIS9_PNS_5ValueENS3_4lessIS9_EENS7_INS3_4pairIKS9_SE_EEEEEE + 6464
2   libcoreir.dylib                     0x00000001107b1d05 _ZN6CoreIR8InstanceC1EPNS_9ModuleDefENSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEEPNS_6ModuleENS3_3mapIS9_PNS_5ValueENS3_4lessIS9_EENS7_INS3_4pairIKS9_SE_EEEEEE + 37
3   libcoreir.dylib                     0x00000001106ed641 _ZN6CoreIR9ModuleDef11addInstanceENSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEPNS_6ModuleENS1_3mapIS7_PNS_5ValueENS1_4lessIS7_EENS5_INS1_4pairIKS7_SC_EEEEEE + 2449
4   libcoreir-c.dylib                   0x00000001104ff1d1 COREModuleDefAddModuleInstance + 2097
5   _ctypes.cpython-36m-darwin.so       0x000000010a4112c7 ffi_call_unix64 + 79
6   ???                                 0x00007ffee764a5c0 0x0 + 140732780553664
[I 19:16:18.256 NotebookApp] KernelRestarter: restarting kernel (1/5)
WARNING:root:kernel cf93a338-335f-408c-8448-be432f4537a1 restarted

I have verified that the exact same code compiles to Verilog and can be successfully run on the IceStick. What is the right way to produce CoreIR JSON from Magma instead of Verilog? 

# Problem 2: What do I do with my CoreIR JSON?

Once I make the CoreIR JSON, what do I do next? I see two options:

1. Get the CoreIR binary to produce the blif and pcf files needed later in the pipeline
2. Convert the CoreIR json to verilog and continue with yosys

What is the right approach? Can you give me an example of the commands to use in that approach?
 I'm not sure if option 1 is possible as, based on the CoreIR help output (see a truncated version below), it doesn't seem like the coreIR binary can produce blif and pcf files.

### CoreIR Help Output:
durst@DN0a22a4ae.SUNet:~/dev/W17-8/magma/notebooks/icestick/build$ coreir
a simple hardware compiler
Usage:
  coreir [OPTION...]

  -h, --help             help
  -v, --verbose          Set verbose
  -i, --input arg        input file: <file>.json
  -o, --output arg       output file: <file>.<json|fir|v|py|dot>
  -p, --passes arg       Run passes in order: '<pass1>,<pass2>,<pass3>,...'
  -e, --load_passes arg  external passes:
                         '<path1.so>,<path2.so>,<path3.so>,...'
  -l, --load_libs arg    external libs:
                         '<path/libname0.so>,<path/libname1.so>,<path/libname2.so>,...'
  -n, --namespaces arg   namespaces to output:
                         '<namespace1>,<namespace2>,<namespace3>,...' (default: global)
  -t, --top arg          top: <namespace>.<modulename>

leonardt, 2018-01-14:
Working on this, looks like there's some issues in magma/coreir that need to be resolved for this path to work. I will let you know once I believe the issues are resolved.

Re: problem 1 you're correct in thinking `m.compile("build/ice_counter", main, output='coreir')`,  the `include_coreir` parameter is no longer needed and should be deprecated. I will open a separate issue for that.

Re: problem 2 My approach would be to use coreir to get verilog, which I would then try to feed through the yosys toolchain in the same process as used for the lattice/verilog backend for magma.

Something like this (again I can test this path out and let you know if it works)
```
coreir --load_libs libcoreir-ice40.dylib -i ice_counter.json -o ice_counter_coreir.v
yosys -q -p 'synth_ice40 -top main -blif ice_counter_coreir.blif' ice_counter_coreir.v
arachne-pnr -q -d 1k -o ice_counter_coreir.txt -p ice_counter_coreir.pcf ice_counter_coreir.blif
icepack ice_counter_coreir.txt ice_counter_coreir.bin

iceprog ice_counter_coreir.bin
```

leonardt, 2018-01-14:
I edited your comment to format it a bit.
Using code block formatting with triple quotes

    ```
    ```

or indenting by 4 spaces (more examples under the code tab at https://guides.github.com/features/mastering-markdown/)

makes it render nicer on the github web interface

leonardt, 2018-01-16:
This should now be working with the `patch-verilog` branch of coreir https://github.com/rdaly525/coreir/pull/356

Once you've confirmed we can close the issue.

leonardt, 2018-05-03:
This is quite stale, and I'm guessing has been resolved. Feel free to re-open if thats the case.

closed 2018-05-03 by leonardt