phanrahan/magma #147: Compiling Magma To CoreIR Module Fails On Connecting Ports That Should Work

David-Durst, 2018-03-16:
The test at https://github.com/David-Durst/aetherling/blob/master/tests/test_up.py#L116 fails with the message:
ERROR: UpSequential: Cannot wire together
  self.I : Bit[5]
  inst0.in : BitIn[5]

This occurs while trying to make the CoreIR C call to connect my UpSequential magma module to the wrapped dehydrate module from CoreIR C++.

Note: for this test to work, you'll need my changes in this branch: https://github.com/David-Durst/magma/tree/aeCoreirIntegration and . I'm going to make the magma pull request once this works. The coreir pull request is https://github.com/rdaly525/coreir/pull/485.

leonardt, 2018-03-16:
Trying to reproduce

On branch `aeCoreirIntegration` for pycoreir, coreir, and magma

Getting 
```
E       AttributeError: 'Context' object has no attribute 'run_passes_namespaced'
```
Made the following change based on your coreir/pycoreir changes
```diff
diff --git a/magma/coreirModuleWrapper.py b/magma/coreirModuleWrapper.py
index 0ec8c8e..eac1a4a 100644
--- a/magma/coreirModuleWrapper.py
+++ b/magma/coreirModuleWrapper.py
@@ -13,5 +13,5 @@ def DefineModuleWrapper(cirb: CoreIRBackend, coreirModule):
 def ModuleFromGeneratorWrapper(cirb: CoreIRBackend, namespace: str, generator: str,
                                dependentNamespaces: list, genargs = {}, modargs = {}):
     moduleToWrap = cirb.context.import_generator(namespace,generator)(**genargs)
-    cirb.context.run_passes_namespaced(["rungenerators"], [namespace] + dependentNamespaces)
-    return DefineModuleWrapper(cirb, moduleToWrap)(**modargs)
\ No newline at end of file
+    cirb.context.run_passes(["rungenerators"], [namespace] + dependentNamespaces)
+    return DefineModuleWrapper(cirb, moduleToWrap)(**modargs)
```

Then I'm getting
```
ERROR: Missing namespace: commonlib

0   libcoreir.dylib                     0x00000001088f36b7 _ZN6CoreIR11PassManager3runENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 2055
1   libcoreir.dylib                     0x000000010870fef6 _ZN6CoreIR7Context9runPassesENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 150
2   libcoreir-c.dylib                   0x00000001086277e7 COREContextRunPasses + 2039
3   _ctypes.cpython-36m-darwin.so       0x0000000105e572c7 ffi_call_unix64 + 79
4   ???                                 0x00007ffeea3a7450 0x0 + 140732828120144
```

David-Durst, 2018-03-16:
I'll push my update to magma in a few min to call the right function.

Is the commonlib dylib in your coreir lib folder? Is the dylib installed?

David-Durst, 2018-03-16:
Pushed updates to magma to use right run_passes.

/usr/local/lib/libcoreir-commonlib.dylib is my installed location of commonlib dylib

leonardt, 2018-03-16:
```
❯ ls /usr/local/lib | grep coreir
libcoreir-aetherlinglib.dylib
libcoreir-c.dylib
libcoreir-cgralib.dylib
libcoreir-commonlib.dylib
libcoreir-ice40.dylib
libcoreir-python.dylib
libcoreir-rtlil.dylib
libcoreir.dylib
```

rdaly525, 2018-03-16:
This is a really strange error. Is there a chance that there are two separate coreir contexts occurring?

leonardt, 2018-03-16:
Okay, got to the error.

I had to make the following change for it to work, will keep it on for now but is there a place where we're supposed to load the library? `run_passes` should probably assume that it's already loaded? Perhaps it could check and load on demand
```
❯ git diff
diff --git a/coreir/context.py b/coreir/context.py
index 480c9a5..b78665d 100644
--- a/coreir/context.py
+++ b/coreir/context.py
@@ -158,6 +158,8 @@ class Context:
         return self.get_lib(lib).generators[name]

     def run_passes(self, passes, namespaces = ["global"]):
+        for namespace in namespaces:
+            self.get_lib(namespace)
         pass_arr = (ct.c_char_p * len(passes))(*(p.encode() for p in passes))
         namespaces_arr = (ct.c_char_p * len(namespaces))(*(n.encode() for n in namespaces))
         return libcoreir_c.COREContextRunPasses(self.context,
```

leonardt, 2018-03-16:
@rdaly525 that's a good hunch, I'll investigate that

leonardt, 2018-03-16:
got it, the coreir simulator is creating a new context, in this case it should be sharing a context with the one used by the generator

rdaly525, 2018-03-16:
Is this inherently a linking issue?

leonardt, 2018-03-16:
Hmm, so making the simulator and the backend share a context didn't quite fix it so I will need to investigate more

leonardt, 2018-03-16:
Ah, aetherling itself is also using a coreir backend (which has its own context)

leonardt, 2018-03-16:
Got it, it was an issue with multiple contexts

David-Durst, 2018-03-16:
Does that fix the commonlib issue? Any idea why I wasn't seeing that?

@leonardt , do you see my wiring issue now?

leonardt, 2018-03-16:
@David-Durst Here's the changes to aetherling I needed to make (pass the backend's context to the simulator)

```diff
diff --git a/aetherling/modules/up.py b/aetherling/modules/up.py
index ad0423e..e75e2f0 100644
--- a/aetherling/modules/up.py
+++ b/aetherling/modules/up.py
@@ -92,4 +92,4 @@ x = ModuleFromGeneratorWrapper(cirb, "aetherlinglib", "dehydrate", {"hydratedTyp
 dehydrate = cirb.context.import_generator("aetherlinglib", "dehydrate")(hydratedType = cirb.get_type(Array(3, Array(5, BitIn)), True))
 cirb.context.give_coreir_module_definition(dehydrate)
 
-"""
\ No newline at end of file
+"""
diff --git a/tests/test_up.py b/tests/test_up.py
index 28dcb32..a3980cd 100644
--- a/tests/test_up.py
+++ b/tests/test_up.py
@@ -107,13 +107,14 @@ def test_up_sequential():
 
     testcircuit = DefineCircuit('TestUpSequential', *args)
 
-    upSequential = UpSequential(CoreIRBackend(c), numElements, inType)
+    coreir_backend = CoreIRBackend(c)
+    upSequential = UpSequential(coreir_backend, numElements, inType)
     wire(upSequential.I, testcircuit.I)
     wire(testcircuit.O, upSequential.O)
 
     EndCircuit()
 
-    sim = CoreIRSimulator(testcircuit, testcircuit.CLK)
+    sim = CoreIRSimulator(testcircuit, testcircuit.CLK, context=coreir_backend.context)
 
     sim.set_value(testcircuit.I, int2seq(testVal, width), scope)
     sim.evaluate()
@@ -169,4 +170,4 @@ sim = CoreIRSimulator(testcircuit, testcircuit.CLK)
 sim.set_value(testcircuit.I, int2seq(testVal, width), scope)
 sim = CoreIRSimulator(testcircuit, testcircuit.CLK)
 simulate(testcircuit, CoreIRSimulator)
-"""
\ No newline at end of file
+"""
```

You'll also need to get the commit 517ca3ea49f366c167c18b5d4be2a79f9a74dab6 from magma (you can merge `coreir-dev` into your `aeCoreirIntegration` to grab it)

leonardt, 2018-03-16:
@David-Durst Your test now gets to
```
2172 >               raise Exception(f"Output {input} of {definition.name} not connected.".format(input))
2173 E               Exception: Output ready of UpSequential not connected.
```
for me, so I think you can take it from there.

I needed commit  517ca3e and this patch to magma
```
❯ git diff
diff --git a/magma/coreirModuleWrapper.py b/magma/coreirModuleWrapper.py
index 641b6e8..98bb1ca 100644
--- a/magma/coreirModuleWrapper.py
+++ b/magma/coreirModuleWrapper.py
@@ -12,6 +12,8 @@ def DefineModuleWrapper(cirb: CoreIRBackend, coreirModule):

 def ModuleFromGeneratorWrapper(cirb: CoreIRBackend, namespace: str, generator: str,
                                dependentNamespaces: list, genargs = {}, modargs = {}):
+    for ns in [namespace] + dependentNamespaces:
+        cirb.context.get_lib(ns)
```

leonardt, 2018-03-16:
@David-Durst Yea that's strange, I'm not sure why either. I couldn't find a place where `commonLib` is loaded before you call `cirb.context.run_passes(["rungenerators"], [namespace] + dependentNamespaces)` inside `ModuleFromGeneratorWrapper`

leonardt, 2018-03-16:
Since `get_lib` caches the libraries, I think it's safe to try and load those dependent libraries before running the passes (since it will just return the cached ones for those already loaded)

leonardt, 2018-03-16:
@David-Durst can you confirm that you can get to `Exception: Output ready of UpSequential not connected.` then close this issue? (As always we can open a subsequent one if you run into a new one)

David-Durst, 2018-03-16:
Yep. I get passed there now. I'd still like a debrief on what the two issues were and why this solved it.

closed 2018-03-16 by David-Durst

leonardt, 2018-03-16:
Trying to use multiple coreir contexts is a recipe for strange behavior, since I think coreir does a lot of caching types, so those cached types won't be shared across contexts and any logic that depends on cached types will not work (I suspect the above issue with wiring was related to the "flip" of the type not matching, even though it did, because the type check was done by checking if the types were literally the same cached type object, Ross may have a better sense of how that type check is performed). 

More generally, we should probably think about where coreir contexts are used in our Python system, and think about the best way to manage the sharing of contexts across system components.

rdaly525, 2018-03-16:
Yeah, lenny described it correctly. It is definitely not valid to do any operation across multiple coreir contexts. LLVM has the same constraint.