phanrahan/magma #155: Issue Using CoreIR Memory Module in Simulator/Interpreter

David-Durst, 2018-04-02:
The test https://github.com/David-Durst/aetherling/blob/master/tests/test_updown.py#L52 uses memory. I don't think this is an issue with my code as I've wired up all the ports (https://github.com/David-Durst/aetherling/blob/master/aetherling/helpers/image_RAM.py#L40) and I'm running the verifyconnectivity pass. Also, I've dumped the unflattened module to JSON and confirmed that everything is wired up there. See https://github.com/David-Durst/aetherling/blob/f7db9ec3a93c7eccb4a0dfa8e300f5aad06e461f/updown_out.json#L460-L470 where inst4, the memory module that is failing, has its raddr, waddr, rdata, wdata, clk, and wen wired up.

However, the interpreter still says something is missing. The issue is at https://github.com/rdaly525/coreir/blob/dev/src/simulator/interpret.cpp#L1045 (I've added some debugging information. That is why my error message below is at line 1049.) Any idea for what's wrong? 

The failure occurs on the first call of the simulator. When initializing the CoreIR python simulator (https://github.com/phanrahan/magma/blob/master/magma/simulator/coreir_simulator.py#L162), the CoreIR python simulator calls reset_circuit on the simulator, which runs exeCombinational for the first time and that fails on its first call to updateMemoryOutput at https://github.com/rdaly525/coreir/blob/dev/src/simulator/interpret.cpp#L1271. 

```
numVertices(g)    = 478
wired node: self.CLK, sequential ? 0, isReceiver ? 0
wired node: inst4, sequential ? 1, isReceiver ? 0
inConns size: 0
inConns instance: inst4
inConns moduleref: Module: mem(depth:2457600, has_init:False, width:24)
  Type: {'clk':coreir.clkIn, 'wdata':BitIn[24], 'waddr':BitIn[22], 'wen':BitIn, 'rdata':Bit[24], 'raddr':BitIn[22]}
  Def? No
inConns type: Instance
Assertion failed: (inConns.size() == 1), function updateMemoryOutput, file interpret.cpp, line 1049.
Abort trap: 6
```

dillonhuff, 2018-04-02:
This is probably a leftover from when the interpreter assumed bulk connections. I'll look into it.

Does the raddr port driven by a single bulk connection or several bitwise connections?

David-Durst, 2018-04-02:
What is a bulk connection vs several bitwise connections? The raddr takes an array of bits (https://github.com/rdaly525/coreir/blob/dev/src/ir/headers/core.hpp#L188).

David-Durst, 2018-04-02:
This is a bulk connection, see https://github.com/David-Durst/aetherling/blob/master/updown_out.json#L466 and https://github.com/David-Durst/aetherling/blob/master/aetherling/helpers/image_RAM.py#L51

dillonhuff, 2018-04-02:
Those links are both dead ends for me. Do they work for you?

David-Durst, 2018-04-02:
@dillonhuff solved this issue for me. It's in another branch in his repo and needs to be pulled into dev.

closed 2018-04-02 by David-Durst