phanrahan/magma #191: Make Wire Throw Exception If Invalid Input

David-Durst, 2018-05-20:
Sometimes I pass in a definition instead of an instance and get an obtuse error like roughly "using an in instead of an out" as the definitions and instances have ports in inverse directions. 

To solve this, I tried to add input verification to wire (https://github.com/phanrahan/magma/blob/master/magma/wire.py#L13) that throws an exception if i or o is not an instance. However, that doesn't work as some things currently calling wire pass in objects that do not have is_instance defined. What are these? How can we add better error messages to wire?

phanrahan, 2018-05-20:
Can you post an example where you get an error.

The wire function mostly just calls the wire method on the object that is passed in.

David-Durst, 2018-05-20:
The checks I added: https://github.com/phanrahan/magma/blob/wiringAssertion/magma/wire.py#L14

This test fail: https://github.com/phanrahan/magma/blob/wiringAssertion/tests/test_wire/test_const.py#L5

So wire can take Int, Instance, and what else should be checked for?

Error message:
```
============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/magma, inifile:
collected 1 item
test_const.py F
tests/test_wire/test_const.py:4 (test_const0)
def test_const0():
        Buf = DeclareCircuit('Buf', "I", In(Bit), "O", Out(Bit))
    
        main = DefineCircuit("main", "O", Out(Bit))
    
        buf = Buf()
    
>       wire(0, buf.I)

test_const.py:12: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../magma/debug.py:24: in wire
    return fn(i, o, debug_info)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

o = 0, i = inst0.I
debug_info = ('/Users/durst/dev/W17-8/magma/tests/test_wire/test_const.py', 12)

    @debug_wire
    def wire(o, i, debug_info):
        if not hasattr(o, "is_instance") or not o.is_instance:
>           raise TypeError("The first input to wire needs to be a circuit instance and is not")
E           TypeError: The first input to wire needs to be a circuit instance and is not

../../magma/wire.py:15: TypeError
                                                          [100%]

=================================== FAILURES ===================================
_________________________________ test_const0 __________________________________

    def test_const0():
        Buf = DeclareCircuit('Buf', "I", In(Bit), "O", Out(Bit))
    
        main = DefineCircuit("main", "O", Out(Bit))
    
        buf = Buf()
    
>       wire(0, buf.I)

test_const.py:12: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../magma/debug.py:24: in wire
    return fn(i, o, debug_info)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

o = 0, i = inst0.I
debug_info = ('/Users/durst/dev/W17-8/magma/tests/test_wire/test_const.py', 12)

    @debug_wire
    def wire(o, i, debug_info):
        if not hasattr(o, "is_instance") or not o.is_instance:
>           raise TypeError("The first input to wire needs to be a circuit instance and is not")
E           TypeError: The first input to wire needs to be a circuit instance and is not

```

leonardt, 2018-05-21:
I like the idea of adding better error messages to wire. As Pat said, we should support wiring 0 and 1 for VCC and GND. That should resolve the errors you posted above. Not sure if there's other non-standard wiring inputs we support, but if we do we should add tests for them

David-Durst, 2018-05-21:
Not adding error message to wire. This would limit the flexibility and wire is supposed to be flexible.

closed 2018-05-21 by David-Durst