phanrahan/magma #192: MANTLE and MANTLE_TARGET

phanrahan, 2018-05-25:
We have two environment variables that control magma and mantle, MANTLE and MANTLE_TARGET.

Currently, if we want to use coreir, we set MANTLE to 'coreir' (similarly for 'verilog'). I believe these values should be used with MANTLE_TARGET.

We also use these variables when we compile to verilog and generate the physical constraints file. In this case, we need to know the vendor to select the correct constraint file.

I suggest we change MANTLE to VENDOR and just use that for the constraint file. And change the logic for loading mantle to just look at MANTLE_TARGET.

phanrahan, 2018-05-25:
This issue came up when I was compiling a magma program for the de0nano with an Altera Cyclone4. I wanted to use the coreir backend.

The same issue should come up if we try to compile the icestick with the coreir backend.

leonardt, 2018-05-25:
Make sense to me, this will require changing quite of a bit of code with
```
os.environ["MANTLE"] = "coreir"
```
I think most of the coreir tests in mantle have this, but I think a simple sed script should be able to take care of this

```
sed -i '' s/os.environ["MANTLE"] = "coreir"/os.environ["MANTLE_TARGET"] = "coreir"/g
```
Note haven't tested that sed command so it might need some escapes (maybe for " and spaces).

phanrahan, 2018-05-25:
I think we should also not access VENDOR in compile. We should pass the vendor into the compile function.

Btw, is there a way to generate a icestick app with a coreir backend?

leonardt, 2018-05-25:
This commit https://github.com/phanrahan/mantle/commit/b4ddf87b56b97c69d29b579fa79a53738e88d05a removes support for `MANTLE` and migrates mantle to use the `MANTLE_TARGET` variable

leonardt, 2018-05-25:
Here's an example compiling blink through the coreir backend, then using the coreir command line tool to generate verilog, which is passed through the standard yosys flow: https://github.com/phanrahan/magmathon/blob/master/notebooks/advanced/coreir.ipynb

Turns out that this works by default because this line in magma: https://github.com/phanrahan/magma/blob/master/magma/compile.py#L82 treats defaults to the `lattice` vendor since we don't set `MANTLE` anymore.

phanrahan, 2018-05-25:
Thanks for the example. That is enough to get me going.

I still think we should change the way MANTLE_TARGET works, right?

phanrahan, 2018-05-25:
We should still support MANTLE_TARGET == 'ice40'.

leonardt, 2018-06-29:
I think this has been resolved with `m.set_mantle_target`, @phanrahan can we close this issue?

closed 2018-06-29 by phanrahan