phanrahan/magma #195: wireclocks

phanrahan, 2018-06-15:
```
class _SISO(Circuit):
        name = _RegisterName('SISO', n, init, has_ce, has_reset)
        IO = ['I', In(Bit), 'O', Out(Bit)] + \
               ClockInterface(has_ce,has_reset)
        @classmethod
        def definition(siso):
            ffs = FFs(n, init=init, has_ce=has_ce, has_reset=has_reset)
            reg = braid(ffs, foldargs={"I":"O"})
            reg(siso.I)
            wire(reg.O, siso.O)
            wireclock(siso, reg)
            wiredefaultclock(siso, reg)
```

Do we really need the wireclock and wiredefaultclock calls?

leonardt, 2018-06-15:
There `coreir` backend calls `wiredefaultclock` when compiling a definition. It does not call `wireclock`. We should standardize what passes are run automatically and document it. It seems like `wireclock` and `wiredefaultclock` could be run before any backend compiles, so we could add a hook in `compile` before calling `<backend>.compile` to run a generic set of passes.

leonardt, 2020-08-07:
Clocks are now wired automatically

closed 2020-08-07 by leonardt