phanrahan/magma #204: Sum of 2 uints decays to Bits

akeley98, 2018-06-26:
If two uint instances are added, I expect that the result should be of type uint, but the actual type is Bits.
```
>>> import magma as m
>>> print(type(m.uint(1, 8) + m.uint(2, 8)))
Out(Bits(8))
```
This means that I get unexpected results if I chain arithmetic operators on uint types (I may get concatenation instead of the expected addition)
```
>>> m.uint(1, 8) + m.uint(2, 8) + m.uint(255, 8)
array([.O[0], .O[1], .O[2], .O[3], .O[4], .O[5], .O[6], .O[7], 1, 1, 1, 1, 1, 1, 1, 1])
>>> 
```

leonardt, 2018-06-27:
Hmm, I think your code snippet is missing a mantle import to reproduce the issue, but I see it with an `import mantle` statement before the `+` call (since mantle needs to be imported for the operator to be overloaded).

This is an issue with how the operator circuits are defined as a generic function (since bits is the most generic). I can update them to preserve the type of the original operator call (what you're requesting), that seems like better behavior.

leonardt, 2018-06-29:
With https://github.com/phanrahan/mantle/pull/77 merged, the behavior is now:

```
>>> import magma as m
>>> import mantle
>>> print(type(m.uint(1, 8) + m.uint(2, 8)))
Out(UInt(8))
```

closed 2018-06-29 by leonardt