phanrahan/magma #205: Mantle LUT doesn't work with PythonSimulator

akeley98, 2018-06-28:
```
import magma as m
import mantle

# Workaround
def make_LUT(*args, **kwargs):
    for i in range(1000):
        try:
            return mantle.LUT(*args, **kwargs)
        except Exception: pass
    return mantle.LUT(*args, **kwargs)

class Doodad(m.Circuit):
    IO = ["O", m.Out(m.Bit)] + m.ClockInterface()
    @classmethod
    def definition(io):
        counter = mantle.Counter(2)
        lut = make_LUT([0,1,1,1], 2)
        m.wire(counter.O[0], lut.I0)
        m.wire(counter.O[1], lut.I1)
        m.wire(lut.O, io.O)
        
from magma.simulator import PythonSimulator

simulator = PythonSimulator(Doodad, clock=Doodad.CLK)
```

Result:
```
Traceback (most recent call last):
  File "lut_fail.py", line 24, in <module>
    simulator = PythonSimulator(Doodad, clock=Doodad.CLK)
  File "/home/david/Documents/aha/magmathon/magma/magma/simulator/python_simulator.py", line 251, in __init__
    self.txfm = flatten(main_circuit)
  File "/home/david/Documents/aha/magmathon/magma/magma/transforms.py", line 206, in flatten
    wire_new_bit(origbit, newbit, Scope(), primitive_map, bit_map, circuit, flattened_circuit)
  File "/home/david/Documents/aha/magmathon/magma/magma/transforms.py", line 170, in wire_new_bit
    newsource = get_new_source(source_qual, primitive_map, old_circuit, new_circuit)
  File "/home/david/Documents/aha/magmathon/magma/magma/transforms.py", line 112, in get_new_source
    raise MagmaTransformException("Failed to collapse bit to primitive. bit={} type={}".format(old_primitive, type(old_primitive)))
magma.transforms.MagmaTransformException: Failed to collapse bit to primitive. bit=inst0 type=LUT2_14(I0: In(Bit), I1: In(Bit), O: Out(Bit))
```

leonardt, 2018-06-28:
Does this work with the coreir simulator? @phanrahan are we planning to revive support for the PythonSimulator? If so, I'll look into this.

akeley98, 2018-06-28:
How do I use the CoreIR simulator? I was following the simulate notebook at https://github.com/phanrahan/magmathon/blob/master/notebooks/advanced/simulate.ipynb

leonardt, 2018-06-28:
Here's one example: https://github.com/phanrahan/mantle/blob/e96d340e07061dfd510b04fcc4f58aa4187c48b1/tests/test_coreir/test_util/test_pop.py#L14-L19