phanrahan/magma #206: Improve wiring user experience

leonardt, 2018-06-29:
We saw a lot of issues related to wiring during the hackathon. Obvious action items are to include source file and line number when outputting a wiring related error or warning.

More generally I think it's important to enumerate a set of test cases that demonstrate the various classes of wiring errors that users can make, and then validate that the errors result in useful error messages.

phanrahan, 2018-08-08:
Some suggestions.

Make sure to print out file and line numbers when calling error and warning.

Use qualified names. For example, I just changed the verilog backend to use the following form when printing out an error.

```
print(f'Warning (verilog): {str(port.inst.defn.name)}.{str(type(port.inst).name)}_{str(port.inst)}.{str(port)} not connected')
```
which results in
```
Warning (verilog): Rescale.Add16_inst22.I0 no connected
```

leonardt, 2018-08-20:
With 6998248, for the following circuit
```
    A = DeclareCircuit('A', "I", In(Bit), "O", Out(Bit))

    main = DefineCircuit("main", "I", In(Bit), "O", Out(Bit))

    a = A()
    wire(main.I, a.O)
```
we get the error message
```
tests/test_wire/test_errors.py:23: Using main.A_inst0.O (an output) as an input
    wire(main.I, a.O)
```
Notice the name `main.A_inst0.O`, still unforunate that we can't capture the `a` variable name yet, but it's better than before (which was just inst0.O).

rsetaluri, 2018-08-23:
Another oddity:

```
magma:WARNING:mantle/coreir/register.py:30: Number of inputs is not equal to the number of outputs, only 9 of the 3 arguments will be wired
magma:ERROR:                    I = mantle.Mux(n)(reg.O, I, io.CE)
```

`Only 9 of 3` seems confusing. Perhaps rephrase.

leonardt, 2018-08-23:
Fixed with 58a9666
```
def test_muliple_outputs_circuit(caplog):
    A = DeclareCircuit('A', "I", In(Bit), "J", In(Bit), "O", Out(Bit), "U", Out(Bit))

    main = DefineCircuit("main", "I", In(Bit), "O", Out(Bit))

    a = A()
    a(main)
```
produces
```
tests/test_wire/test_errors.py:63: Number of inputs is not equal to the number of outputs, expected 2 inputs, got 1. Only 1 will be wired.
     a(main)
```
instead of (old output)
```
tests/test_wire/test_errors.py:63: Number of inputs is not equal to the number of outputs, only 2 of the 1 arguments will be wired
    a(main)
```

leonardt, 2018-08-23:
I'm going to close this for now since we covered a bunch of ground here. Let's reopen issues as they come (keep them coming!)

closed 2018-08-23 by leonardt