phanrahan/magma #213: Connection error in generating verilog from json

adamdai, 2018-07-13:
I am using the following magma/aetherling code to output a `json` module:
```
from aetherling.modules.linebuffer import *
from aetherling.modules.reduce import ReduceSequential, ReduceParallel, renameCircuitForReduce
from magma import *
from magma.clock import *
from magma.backend.coreir_ import CoreIRBackend
from magma.bitutils import *
from coreir.context import *
from magma.simulator.coreir_simulator import CoreIRSimulator
import coreir
from magma.scope import Scope
from mantle.coreir import DefineCoreirConst
from mantle import CounterModM, Decode, SIPO
from magma.frontend.coreir_ import GetCoreIRModule
from mantle.coreir.arith import *

c = coreir.Context()
cirb = CoreIRBackend(c)
scope = Scope()

# Line Buffer interface
inType = Array(1, Array(1, Array(8, BitIn)))
outType = Array(2, Array(2, Array(8, Out(Bit))))
imgType = Array(4, Array(4, Array(8, Out(Bit))))

# Reduce interface
inType2 = In(Array(4, Array(8, BitIn)))
outType2 = Out(Array(8, Bit))

args = ['I', inType, 'O', outType2] + ClockInterface(False, False)

testcircuit = DefineCircuit('lb1_3_Test', *args)

# Line buffer declaration
lb = Linebuffer(cirb, inType, outType, imgType, False)
wire(lb.I, testcircuit.I)
wire(1, lb.wen)

# # Reduce declaration
reducePar = ReduceParallel(cirb, 4, renameCircuitForReduce(DefineAdd(8)))
coreirConst = DefineCoreirConst(8, 0)()
wire(reducePar.I.data[0], lb.out[0][0])
wire(reducePar.I.data[1], lb.out[0][1])
wire(reducePar.I.data[2], lb.out[1][0])
wire(reducePar.I.data[3], lb.out[1][1])
wire(reducePar.I.identity, coreirConst.out)
wire(testcircuit.O, reducePar.out)

EndCircuit()

module = GetCoreIRModule(cirb, testcircuit)
module.save_to_file("stencil.json")
```
which I am then trying to convert to verilog with the following command:
```
coreir -i stencil.json -o stencil.v -p 'rungenerators,wireclocks-coreir,verifyconnectivity-noclkrst,flattentypes,flatten,verifyconnectivity-noclkrst,deletedeadinstances' -n 'aetherlinglib,commonlib,mantle,coreir,global' --load_libs 'libcoreir-aetherlinglib.so,libcoreir-commonlib.so'
```
which produces the following error:
```
ERROR: {reduceParallel}.self.in.identity Is not connected
with params=(numInputs:4, operator:global.renamedForReduce_opAdd8_I0-In_Bits_8__-I1-In_Bits_8__-O-Out_Bits_8___)

I AM DYING!
coreir: context.cpp:112: void CoreIR::Context::die(): Assertion `0' failed.
Aborted (core dumped)
```
Inspecting the `json` file shows that the `identity` input is connected so we weren't sure what is causing this. The code and the generated `json` can also be found in https://github.com/adamdai/magmacam/tree/master/aetherling

@David-Durst 
@leonardt 
@rdaly525

leonardt, 2018-07-13:
So I ran this
```shell
coreir -i stencil.json -p rungenerators -o stencil_generated.json -n 'aetherlinglib,commonlib,mantle,coreir,global' --load_libs 'libcoreir-aetherlinglib.dylib,libcoreir-commonlib.dylib'
```

and looked in side 

to find
```json
...
      "reduceParallel":{
        "typegen":"aetherlinglib.reduceParallel_type",
        "genparams":{"numInputs":"Int", "operator":"Module"},
        "modules":[
          [{"numInputs":["Int",4], "operator":["Module","global.renamedForReduce_opAdd8_I0-In_Bits_8__-I1-In_Bits_8__-O-Out_Bits_8___"]},{
            "type":["Record",[
              ["in",["Record",[["data",["Array",4,["Array",8,"BitIn"]]],["identity",["Array",8,"BitIn"]]]]],
              ["out",["Array",8,"Bit"]]
            ]],
            "instances":{
              "reducer":{
                "genref":"aetherlinglib.reduceParallelPower2Inputs",
                "genargs":{"numInputs":["Int",4], "operator":["Module","global.renamedForReduce_opAdd8_I0-In_Bits_8__-I1-In_Bits_8__-O-Out_Bits_8___"]}
              }
            },
            "connections":[
              ["self.in.data.0","reducer.in.0"],
              ["self.in.data.1","reducer.in.1"],
              ["self.in.data.2","reducer.in.2"],
              ["self.in.data.3","reducer.in.3"],
              ["self.out","reducer.out"]
            ]
          }]
        ]
      },
...
```

Seems like the issue is that the reduceParallel generator is generating a module with identity not connected to anything. Technically this could be okay? Since an unwired input will just be dangling, but I'm not sure if this is the expected behavior, and it seems like verify connectivity is being strict about this?

David-Durst, 2018-07-13:
What is happening is that reduceParallel is generating a module with an input for self.in.identity. https://github.com/adamdai/magmacam/blob/master/aetherling/test_linebuffer.py#L47 shows Adam connecting this to the output of a constant. Additionally, https://github.com/adamdai/magmacam/blob/master/aetherling/stencil.json#L64 and https://github.com/adamdai/magmacam/blob/master/aetherling/stencil.json#L74 show that this constant module exists in the CoreIR json produced by magma before the CoreIR passes are run. However, this constant disappears after the runGenerators pass. Why is this constant disappearing?

leonardt, 2018-07-13:
It looks like the constant is still there in the output (inst2)
```
      "lb1_3_Test":{
        "type":["Record",[
          ["I",["Array",1,["Array",1,["Array",8,"BitIn"]]]],
          ["O",["Array",8,"Bit"]],
          ["CLK",["Named","coreir.clkIn"]]
        ]],
        "instances":{
          "bit_const_VCC":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",true]}
          },
          "inst0":{
            "genref":"commonlib.linebuffer",
            "genargs":{"has_valid":["Bool",false], "image_type":["CoreIRType",["Array",4,["Array",4,["Array",8,"BitIn"]]]], "input_type":["CoreIRType",["Array",1,["Array",1,["Array",8,"BitIn"]]]], "is_last_lb":["Bool",true], "output_type":["CoreIRType",["Array",2,["Array",2,["Array",8,"Bit"]]]]}
          },
          "inst1":{
            "genref":"aetherlinglib.reduceParallel",
            "genargs":{"numInputs":["Int",4], "operator":["Module","global.renamedForReduce_opAdd8_I0-In_Bits_8__-I1-In_Bits_8__-O-Out_Bits_8___"]}
          },
          "inst2":{
            "modref":"global.coreir_const80"
          }
        },
        "connections":[
          ["inst0.wen","bit_const_VCC.out"],
          ["self.I","inst0.in"],
          ["inst1.in.data.0","inst0.out.0.0"],
          ["inst1.in.data.1","inst0.out.0.1"],
          ["inst1.in.data.2","inst0.out.1.0"],
          ["inst1.in.data.3","inst0.out.1.1"],
          ["inst2.out","inst1.in.identity"],
          ["self.O","inst1.out"]
        ]
      },
```
I think the issue is inside the reduceParallel module which has an unwired port in its definition (just viewing the definition in it's own scope)

leonardt, 2018-07-13:
(The above snippet was copied from `stencil_generated.json.txt`)

David-Durst, 2018-07-13:
@leonardt you're right. In the event that a power 2 inputs is provided, identity is left dangling. I screwed up a constant in my unit tests and so even though this case was supposed to be tested for, it wasn't. I've fixed it and added a test for it with https://github.com/rdaly525/coreir/pull/577. I'll close this when that gets accepted.

leonardt, 2018-08-27:
coreir#577 was accepted, closing.

closed 2018-08-27 by leonardt