phanrahan/magma #217: One unused GPIO causes all GPIOs to be seen as undriven [IceStick]

adamdai, 2018-07-17:
I'm not sure if this is something people already aware of, but I've noticed while debugging my ice40 programs that I often declare a set of GPIOs then only wire a subset of them up. While this is an easy thing to fix, it seems that leaving any GPIOs unwired causes all other GPIOs to be seen as unwired as well. 

To replicate this issue, an Icestick and logic analyzer are need. If the following code is compiled and uploaded to the Icestick, and the pins J3[0] and J3[1] are connected to the logic analyzer, both signals should show up no problem.
```
from magma import *
set_mantle_target("ice40")
from mantle import*
from loam.boards.icestick import IceStick

icestick = IceStick()
icestick.Clock.on()
for i in range(2):
    icestick.J3[i].output().on()

main = icestick.main()

clock = CounterModM(103, 8)
baud = clock.COUT

wire(main.CLKIN, main.J3[0])
wire(baud, main.J3[1])
```
Now, if you comment out just one of the `wire` calls are the bottom, say `#wire(baud, main.J3[1])`, then now compiling with yosys gives the warnings:
```
Warning: Wire main.\J3 [1] is used but has no driver.
Warning: Wire main.\J3 [0] is used but has no driver.
```
even though `main.J3[0]` should still have a driver, and if you look at the logic analyzer output, you see that indeed none of the GPIOs are being driven. Not sure if this is a priority but just wanted to point it out since it would be nice if magma could just ignore unwired GPIOs without interfering with the other ones.

phanrahan, 2018-07-20:
Yes, there is an issue with using subsets of the bits in an bits array. In particular, the current implementation does not allow you to have sparse indices. You need to use a continuous range of indices from 0 to n.

The above code should work if you comment out the J1[1], but will not work if you comment out J10].

One workaround for this is to `rename` the GPIO.

```
icestick.J3[i].output().rename(f'J1{i}).on()
```

The reason this doesn't work is that yosys and arachne-pnr don't handle subsets of arrays.

adamdai, 2018-07-23:
Okay I see, I think it actually breaks if you comment out either J1[0] or J1[1], but I understand that the source of the issue lies underneath magma so there might not be an easy fix

closed 2018-07-23 by adamdai