phanrahan/magma #218: Basic Tests Requiring CounterModM Fail

David-Durst, 2018-07-20:
I'm on the latest master of everything. Make cleaned, removed coreir binaries manually, reinstalled them.

Running https://github.com/David-Durst/aetherling/blob/master/tests/test_up.py#L39 produces the following error:

```
(aetherling) durst@DN0a22a1d6:~/dev/W17-8/aetherling$ pytest -s tests/test_up.py
=========================================================================================================== test session starts ============================================================================================================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 3 items

tests/test_up.py In Run Generators
Done running generators
Starting topological sort
topo_order.size() = 3
numVertices(g)    = 3
.In Run Generators
Done running generators
Starting topological sort
topo_order.size() = 6
numVertices(g)    = 20
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file ../../include/coreir/ir/casting/casting.h, line 249.
Abort trap: 6
```

leonardt, 2018-07-20:
Here's the backtrace from lldb
```
❯ lldb -- python tests/test_up.py
(lldb) target create "python"
Current executable set to 'python' (x86_64).
(lldb) settings set -- target.run-args  "tests/test_up.py"
(lldb) r
Process 61823 launched: '/Users/lenny/miniconda3/bin/python' (x86_64)
In Run Generators
Done running generators
Starting topological sort
topo_order.size() = 6
numVertices(g)    = 20
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file ../../include/coreir/ir/casting/casting.h, line 249.
Process 61823 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff73eefb6e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff73eefb6e <+10>: jae    0x7fff73eefb78            ; <+20>
    0x7fff73eefb70 <+12>: movq   %rax, %rdi
    0x7fff73eefb73 <+15>: jmp    0x7fff73ee6b00            ; cerror_nocancel
    0x7fff73eefb78 <+20>: retq
Target 0: (python) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff73eefb6e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff740ba080 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff73e4b1ae libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff73e131ac libsystem_c.dylib`__assert_rtn + 320
    frame #4: 0x0000000104e01267 libcoreir.dylib`CoreIR::cast_retty<CoreIR::Select, CoreIR::Wireable*>::ret_type CoreIR::cast<CoreIR::Select, CoreIR::Wireable>(CoreIR::Wireable*) + 103
    frame #5: 0x0000000105273415 libcoreir.dylib`CoreIR::toSelect(CoreIR::Wireable*) + 21
    frame #6: 0x0000000105272e98 libcoreir.dylib`CoreIR::SimulatorState::getValue(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 88
    frame #7: 0x000000010528b6f7 libcoreir.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 71
    frame #8: 0x000000010528c3a9 libcoreir.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 3321
    frame #9: 0x000000010528b671 libcoreir.dylib`CoreIR::SimulatorState::getValueByOriginalName(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) + 65
    frame #10: 0x0000000104c21121 libcoreir-c.dylib`CORESimGetValueByOriginalName + 97
    frame #11: 0x0000000103eea884 libffi.6.dylib`ffi_call_unix64 + 76
    frame #12: 0x0000000103ee9e8b libffi.6.dylib`ffi_call + 939
    frame #13: 0x0000000103ed3cf8 _ctypes.cpython-36m-darwin.so`_ctypes_callproc + 552
    frame #14: 0x0000000103ece222 _ctypes.cpython-36m-darwin.so`PyCFuncPtr_call + 242
    frame #15: 0x0000000100009341 python`_PyObject_FastCallDict + 177
    frame #16: 0x0000000100161948 python`call_function + 392
    frame #17: 0x000000010015f54a python`_PyEval_EvalFrameDefault + 47098
    frame #18: 0x0000000100152b89 python`_PyEval_EvalCodeWithName + 425
    frame #19: 0x0000000100161cba python`fast_function + 362
    frame #20: 0x00000001001618ac python`call_function + 236
    frame #21: 0x000000010015f54a python`_PyEval_EvalFrameDefault + 47098
    frame #22: 0x0000000100161c0c python`fast_function + 188
    frame #23: 0x00000001001618ac python`call_function + 236
    frame #24: 0x000000010015f54a python`_PyEval_EvalFrameDefault + 47098
    frame #25: 0x0000000100161c0c python`fast_function + 188
    frame #26: 0x00000001001618ac python`call_function + 236
    frame #27: 0x000000010015f54a python`_PyEval_EvalFrameDefault + 47098
    frame #28: 0x0000000100152b89 python`_PyEval_EvalCodeWithName + 425
    frame #29: 0x00000001001ab0dc python`PyRun_FileExFlags + 252
    frame #30: 0x00000001001aa83e python`PyRun_SimpleFileExFlags + 366
    frame #31: 0x00000001001d1156 python`Py_Main + 3734
    frame #32: 0x0000000100001469 python`main + 313
    frame #33: 0x00007fff73d9f015 libdyld.dylib`start + 1
    frame #34: 0x00007fff73d9f015 libdyld.dylib`start + 1
(lldb)
```

leonardt, 2018-07-20:
Fixed

closed 2018-07-20 by leonardt