phanrahan/magma #222: Test out global wires

rsetaluri, 2018-07-27:
We want to experiment with some construct to have global wires and contexts so that we don't have to always wire through intermediate modules, or muddy interfaces with a bunch of common things.

See https://docs.google.com/document/d/1AmKWV8YKTjCQuqzOFUsVFZbgZjr9Q_tCDqCUgqXCThI/edit for more details

rsetaluri, 2018-07-27:
Started experimenting with this here: https://github.com/phanrahan/magma/blob/global-wire-experiment/magma/experimental/global_wire_experiment/test.py

phanrahan, 2018-07-28:
Take a look at #208. More generally, we should clean up clock wiring.

phanrahan, 2018-07-28:
I looked at your example.  Nice.

My main comment is whether we want to use lexical scoping for global wires. I can't think of a case right now, but I can imagine that could be constraining.

This is important enough that we may want to introduce a type qualifier Global(T).

leonardt, 2018-07-28:
Yea, I was thinking about scoping as well. There is a notion of referring to the scope in which you are instantiated (dynamic scoping?). For example, wire the clock of the module that I'm instanced in to this register. (I use `parent` to refer to the CLK port of the module I'm instantiated in. 
```
wire(parent.CLK, reg.CLK)
```
More generally, this seems a bit like dynamic scoping (using the scope in which you are invoked) versus lexical scoping (using the scope in which you are defined).

I think `globals` being lexically scoped makes sense, since any definition context has access to the same (lexical/dynamic) global scope. For more fine grained scoping rules (e.g. multiple clock domains organized by module hierarchy), we might consider using dynamic scoping (inheriting the scope in which you are instanced in). This would also allow us to use the `with` context manager pattern for setting variables.

```
with CLK1 as CLK:
    inst = CircuitThatUsesCLK()
with CLK2 as CLK:
    inst = CircuitThatUsesCLK()
```

Global(T) sounds interesting. I like the idea of making this explicit (like the `global` keyword in Python). It seems like this would allow us to verify that only a global signal is hooked up to a port that is declared to be global.