phanrahan/magma #226: Simulator interface

phanrahan, 2018-07-31:
The python simulator often returns lists of bools. This can happen with numeric operators.

I propose that we change interface to return the following types:

Bit = Bool
Array(n,Bit) = [bool]
Bits(n) = uint
UInt(n) = uint
SInt(n)= int

when returning ints we should make sure to respect the width of the circuit definition.

Same should apply to input values.

We could run conversion functions to get to the proper type.

leonardt, 2018-08-21:
As of ec6987733d37de3b133526f8d75dfaa19539c3fc, the simulator now implements the above interface with the following additions.

BitVectors can be passedfor Array, Bits, UInt, SInt and it will be converted using the conversion as_bool_list()

closed 2018-08-21 by leonardt

leonardt, 2018-08-21:
Can I propose we return BitVector for `Bits(n)`?

phanrahan, 2018-08-21:
What is the reason to return BitVector? I was suggesting we stick with standard python types.

leonardt, 2018-08-21:
There's quite a bit of code in the simulator internals that leverage the fact that it uses a list of bits. But that was more related to the get_value/set_value semantics. Perhaps it's worth reverting these changes to only have this interface apply to the __call__ syntax? Then the set_value and get_value semantics could remain the same (list of bools)