phanrahan/magma #228: Create example with @circuit.combinational that can emulate a verilog FSM

rdaly525, 2018-07-31:
You should be able to nicely emulate something like the following Verilog FSM (which I wrote a long time ago for Rigel)
```

`define REG(clk, r, init, in) \
    always @(posedge clk or negedge rst_n) begin \
        if (!rst_n) r <= (init); \
        else r <= (in); \
    end

    localparam IDLE=2'h0, HBLANK=2'h1, HACT=2'h2;
    reg [10:0] pix_cnt_n;
    reg [10:0] pix_cnt;

    reg [1:0] pix_ns;
    reg [1:0] pix_cs;

    assign pixel_valid = (pix_cs == HACT);
    always @(*) begin
        case(pix_cs)
            IDLE : begin
                pix_ns = frameValid ? HBLANK : IDLE;
                pix_cnt_n = 0;
            end
            HBLANK : begin
                pix_ns = real_href ? HACT : HBLANK ;
                pix_cnt_n = real_href ? `PIX_PER_LINE : 0;
            end
            HACT : begin
                pix_ns = (pix_cnt == 1) ? HBLANK : HACT ;
                pix_cnt_n = pix_cnt - 1'b1;
            end
            default : begin
                pix_ns = IDLE ;
                pix_cnt_n = 0;
            end
        endcase

    end
    `REG(pclk, pix_cs, IDLE, pix_ns)
    `REG(pclk, pix_cnt, 0, pix_cnt_n)
```

leonardt, 2018-08-02:
Here's a first pass at the example: https://github.com/phanrahan/mantle/tree/master/examples/example/fsm

Any feedback? I think it comes out pretty clean. I did a basic co-simulation of the two implementations and they pass the same test. One minor change as I changed the `rst_n` to `rst` (posedge) because they're some wonkiness I need to figure out to pass the posedge/negedge parameter from mantle down to coreir.

rdaly525, 2018-08-02:
This looks really good! I think this should be an easily findable example. Honestly my main qualm is python's weird version of ternary expressions. Perhaps a simple m.ternary(cond,true_expr,false_expr) might look better. 

In terms of the example, perhaps even having the def fsm_logic within the definition would make it look even more similar.

This is a good example to go in the "tutorial for verilog people"

phanrahan, 2018-08-02:
I like the C cond ? tvalue : fvalue syntax better than the python tvalue if cond else fvalue. But I recommend we be pythonic when we are in python.

leonardt, 2018-08-03:
I also agree that we should stick with python syntax, also there's an order of evaluation issue here
```
>>> def ite(cond, t, f):
...     return t if cond else f
...
>>> ite(1, print(True), print(False))
True
False
>>> print(True) if 1 else print(False)
True
>>>
```
But I don't think that comes up with magma code since we just rewire the ast and all the hardware expressions get evaluated and the result is muxed based on select, so actually the `ite` version may actually more closely match the semantics.

However, the `ternary` function is trivial to write, so if a user prefers it, they can add a definition in their code. We can revisit this if it there's enough demand from users for it being built in.

I'm going to close this for now since the above example has been implemented and added. I'll plan to add a version to the magmathon repo as a notebook to be the "high-profile" example.

leonardt, 2018-08-03:
(Already opened an issue for this: https://github.com/phanrahan/magmathon/issues/20)

closed 2018-08-03 by leonardt