phanrahan/magma #229: Emulate the variety of verilog case statements in Magma

rdaly525, 2018-07-31:
In verilog and system verilog you have some useful types of case statements such as casez, priorty, and unique.

example from this paper http://shdl.co/papers/2005-SNUG-paper_SystemVerilog_unique_and_priority.pdf 

```
always @* begin
  // set outputs to defaults for when there is no interrupt
  casez (IRQ) // use don’t care bits when evaluating IRQ
    4’b???1: begin // test if IRQ bit 0 is set, ignore other bits 
      // process interrupt 0
    end
    4’b??1?: begin // test if IRQ bit 1 is set, ignore other bits
      // process interrupt 1
    end
    4’b?1??: begin // test if IRQ bit 2 is set, ignore other bits
                   // process interrupt 2
    end
    4’b1???: begin // test if IRQ bit 3 is set, ignore other bits
                   // process interrupt 3
    end 
  endcase
end

```

phanrahan, 2018-07-31:
Remember, in python, there are no case / switch statements. Everything will be done with if-elif-else, 

I think it is better that way. case statements are always problematical in any language. I sent out a PEP reference that proposed to add a switch statement to python, which Guido eventually vetoed.

rdaly525, 2018-07-31:
I understand there are no case statements in python, but the important behavior here that is useful is the wildcard matching for bits. Again, approaching this from a "How can I emulate this productive verilog construct", I think it would be good to be able to do something like this in magma or a small DSL/sugar built on top of magma.

phanrahan, 2018-07-31:
This last example would translate into python as:

```
def process_irq(irq: m.Bits(4)):
    if irq[0]:  # test if IRQ bit 0 is set, ignore other bits 
           # process interrupt 0
    elif irq[1]: # test if IRQ bit 1 is set, ignore other bits
           # process interrupt 1
    elif irq[2]: begin # test if IRQ bit 2 is set, ignore other bits
           # process interrupt 2
    elif irq[3]: # test if IRQ bit 3 is set, ignore other bits
           # process interrupt 3
```

I am not sure how we would handle z or x, since these values are not supported in the synthesized hardware.

rdaly525, 2018-07-31:
The issue is that that would likely result in a serial chain of muxes rather than a single one-hot mux which is the intention of the verilog code. 

We should not handle z or x as those are in the simulator

In addition a slightly more complex pattern matching 

```
case(IRQ)
  4'b1?00: //do somehting 
   ///
endcase

```
would translate into something like:
```
if irq[3] & ~irq[1] & ~irq[0]:
   //do somehting
```

Which is not as concise.
I have seen some pretty complex pattern matching case statements before in which the verilog syntax made it very readable.

leonardt, 2018-07-31:
We could add some sugar for pattern matching like
```
if match(irq, "4b1?00"):
    ...
```
Re: generating the serial muxes, I think this an optimization concern rather than a syntax level concerns. Given a serial chain of muxes and all the conditions specified in a certain mattern (e.g. with a match construct) I think we could generate a one-hot equivalent, I don't see anything about the casez syntax that enables the mux optimization over the if/else syntax. It seems the casez syntax is mainly focused on the notion of don't care.

phanrahan, 2018-07-31:
The reason they add these special case constructs is that the compiler cannot know what possible values the variable can have. For example, it cannot know that the irq is one-hot encoded and only a single bit can be on. As a result, it cannot make that assumption, and the code is not efficient.

Chisel handles this by introducing special muxes for these cases.

```
MuxCase(default, Array(c1 -> a, c2 -> b, ...))
```

https://github.com/freechipsproject/chisel3/wiki/Muxes-and-Input-Selection

phanrahan, 2018-07-31:
@leonardt  match(irq, "4b1?00") would be a good candidate for a macro.

THofstee, 2018-08-01:
@rdaly525 that's not a one hot mux, that's a priority encoder.

rdaly525, 2018-08-01:
It is intended to be a one-hot mux. Adding in the systemverilog 'unique' keyowrd would make it even more explicit.

A priority encoder would be something more like:
```
  casez (IRQ) // use don’t care bits when evaluating IRQ
    4’b???1: begin // test if IRQ bit 0 is set, ignore other bits 
      // process interrupt 0
    end
    4’b??10: begin // test if IRQ bit 1 is set, ignore other bits
      // process interrupt 1
    end
    4’b?100: begin // test if IRQ bit 2 is set, ignore other bits
                   // process interrupt 2
    end
    4’b1000: begin // test if IRQ bit 3 is set, ignore other bits
                   // process interrupt 3
    end 
  endcase
```

THofstee, 2018-08-01:
No, it is a priority encoder. A one hot mux would look like this:

```
unique case (IRQ)
    4'b0001: begin ... end
    4'b0010: begin ... end
    4'b0100: begin ... end
    4'b1000: begin ... end
endcase
```

The document you linked clearly states it is an alternate way of modeling the interrupt encoder in example 2, which has the following behavior:

> In software, the IRQ interrupt bits will be tested one at a time, in the order in which the bits are
listed in the code. As soon as one of the bits tests as true, no other bits are tested. If no bits are set,
then no action is taken.

phanrahan, 2018-08-01:
I read the Sutherland paper on unique and priority. It does explain very well the pitfalls in using sequential behavioral code for synthesis.  We need to make sure we are smart about doing the translation from python to verilog. Proposals welcome!

phanrahan, 2018-08-01:
One more comment. @leonardt's match can be easily written as a python function that returns a Bit if the match is successful. It is a good example of what is possible in Magma.