phanrahan/magma #233: Cast Integers To Constants When Input To Wire

David-Durst, 2018-08-02:
I can do the following:
```python
wire(1, instance.O)
```
But I can't do the following:
```python
x = 255
wire(x, instance.O)
```
This is unintuitive.

I want the feature for the following use case in my linebuffer:
```python
wire(cls.CE &
                 (valid_counter.O < max(used_coordinates) - origin), valid_counter.CE)
```
In this case, I want to be able to automatically cast the int from `max(used_coordinates) - origin` to a constant and wire its output up to the `<` comparator.

David-Durst, 2018-08-03:
Maybe I'm missing something, I thought the below was supposed to work? It's not for me:
```python
wire(enable(bit(cls.CE) & 1), valid_counter.CE)
```

leonardt, 2018-08-03:
I think the idea was that you were requried to use `bits` to convert the integer to a magma value. This is confused (as you pointed out) by the fact that 1/0 work for Bit values.

@phanrahan the options for resolution here seem to be
* (a) do implicit type conversion of integers when using operators or wiring to magma values
* (b) remove the implicit type conversion for 1/0 (and I'm guessing True/False) for working with magma Bit values

leonardt, 2018-08-03:
Actually, from your example, it seems like 1 works for a non-Bit value? That is quite confusing. We should either always do the conversion implicitly, or always require an explicit conversion.

phanrahan, 2018-08-03:
I vote for doing implicit type conversion when assigning to Bit, Bits, UInt, SInt. In all cases, the type conversion is well-defined. @priyanka also wanted this feature.

phanrahan, 2018-08-09:
I implemented a version of this feature. @David-Durst, try it out.

David-Durst, 2018-08-10:
@phanrahan I can't find the commit. Which branch is it on? Also, how does this handle bit widths? If do `wire(6, def.O)` and `def.O` is 32 bits long, will 6 be a 32 bit constant generator?

phanrahan, 2018-08-10:
@David-Durst since this was a simple change, the PR has already been merged into the main branch.

It work as you expect,
```
m.wire(1,m.O) # type(m.O) == m.In(m.Bits(32))
```
is the same as
```
m.wire(m.bits(1,32), m.O
```

David-Durst, 2018-09-11:
I don't think this works. https://github.com/David-Durst/aetherling/blob/266334eca21da1e8085aa3649b866761c95349b7/aetherling/modules/delayed_buffer.py#L61 fails with:
{NotImplementedError}Cannot get_length of <class 'int'>

And `point_in_emitting_period.out` is an array of Uints, so it should be divisible.

leonardt, 2018-09-12:
Question, should we support promoting widths of magma values for operators? For example,
```
a = m.UInt(1)
b = m.UInt(2)

c = a + b
```
Right now this will error since the widths don't match.  Using similar logic for integers, it seems like we could do an implicit promotion here, so the result `c` is the width of the larger element.

leonardt, 2018-09-12:
Closing this issue as it seems to have basic support, see these tests: https://github.com/phanrahan/magma/blob/cc09e3e3a69b4703000a9debbe013c912bbdf925/tests/test_wire/test_const.py#L1-L50

They check wiring 1, 0 to Bit and wiring 1 to Bits, UInt, SInt.

For any future problems related to this feature, let's open up separate issues.

closed 2018-09-12 by leonardt