phanrahan/magma #236: Two Modules That Use _Mux2 Crash When Trying To Simulate In CoreIR

David-Durst, 2018-08-03:
(Note: this issue depends on https://github.com/phanrahan/magma/pull/235)

In the test at https://github.com/David-Durst/aetherling/blob/02231aa19c9b92baf85bfd83dcd76a74e26657ae/tests/test_native_line_buffer.py#L63-L82, I'm creating two magma instances in the same definition. 

Each has its own _Mux2. You can see that in the files outputted when I create the modules separately:

1. tests where I create them separately: https://github.com/David-Durst/aetherling/blob/02231aa19c9b92baf85bfd83dcd76a74e26657ae/tests/test_native_line_buffer.py#L31-L61
1. CoreIR JSON Output File 1: [multiple_sipo.json.txt](https://github.com/phanrahan/magma/files/2256096/multiple_sipo.json.txt)
2. CoreIR JSON Output File 2: [sized_counter_modm.json.txt](https://github.com/phanrahan/magma/files/2256095/sized_counter_modm.json.txt)

When I run the test, I get a _Mux2 namespace collision. I should not get a namespace collision. The error is below.

```Shell
Testing started at 10:26 PM ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --target test_native_line_buffer.py::test_multiple_sipo_and_counter -- -s
Launching pytest with arguments -s test_native_line_buffer.py::test_multiple_sipo_and_counter in /Users/durst/dev/W17-8/aetherling/tests

============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 1 item
test_native_line_buffer.py In Run Generators
Done running generators
ERROR: _Mux2 already exists in global

0   libcoreir.dylib                     0x000000010c66c211 _ZN6CoreIR9Namespace13newModuleDeclENSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEPNS_4TypeENS1_3mapIS7_PNS_9ValueTypeENS1_4lessIS7_EENS5_INS1_4pairIKS7_SC_EEEEEE + 145
1   libcoreir-c.dylib                   0x000000010c3afd18 CORENewModule + 2296
2   libffi.6.dylib                      0x0000000109f7c884 ffi_call_unix64 + 76
3   ???                                 0x00007ffee69ae7f0 0x0 + 140732767332336

Process finished with exit code 1

```

David-Durst, 2018-08-03:
Interestingly, this passes https://github.com/David-Durst/aetherling/blob/43246322bdf7e3ddc65db8b2d17fdc6c834a6a46/tests/test_native_line_buffer.py#L79-L97.

Even though map is not duplicating anything, something about it is bypassing the module anti-deduplication in Magma or CoreIR.

David-Durst, 2018-08-03:
Responding to Ross's question in https://github.com/David-Durst/aetherling/issues/12, I'm not duplicating the CoreIR module in the map at https://github.com/rdaly525/coreir/blob/master/src/libs/aetherlinglib/aeMap.h#L87. I'm instancing it.

leonardt, 2018-08-03:
Fixed by 96e8f900cbe710baf7a89060ddc1692d573ae600

Basically, the backend was not saving already compile modules across multiple calls to compile. @David-Durst can you check it out and close this if that works for you?

David-Durst, 2018-08-03:
This fixes my issue. Thanks!

David-Durst, 2018-08-03:
Actually, it appears the issue is still there in the simulator. The tests work when creating the CoreIR json file, but it's the same error for https://github.com/David-Durst/aetherling/blob/7ce4df31d94bedfa2b151cfd375bfac895ba44c2/tests/test_native_line_buffer.py#L108-L128 when running the simulator:
```shell
Testing started at 12:39 AM ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --target test_native_line_buffer.py::test_multiple_sipo_and_counter -- -s
Launching pytest with arguments -s test_native_line_buffer.py::test_multiple_sipo_and_counter in /Users/durst/dev/W17-8/aetherling/tests

============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 1 item
test_native_line_buffer.py In Run Generators
Done running generators
ERROR: _Mux2 already exists in global

0   libcoreir.dylib                     0x000000010d2b2211 _ZN6CoreIR9Namespace13newModuleDeclENSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEPNS_4TypeENS1_3mapIS7_PNS_9ValueTypeENS1_4lessIS7_EENS5_INS1_4pairIKS7_SC_EEEEEE + 145
1   libcoreir-c.dylib                   0x000000010cff5d18 CORENewModule + 2296
2   libffi.6.dylib                      0x000000010abc2884 ffi_call_unix64 + 76
3   ???                                 0x00007ffee5d683b0 0x0 + 140732754461616

Process finished with exit code 1
```

leonardt, 2018-08-03:
Should be fixed by 0a08d02

Was a similar issue, decided to have the backend cache the modules already created by context (in this case, a new backend was getting created with an old context, so there was a fresh set of modules (nothing defined) even though in a previous backend modules had been defined). Now each coreir context has an associated set of already defined modules.

leonardt, 2018-08-03:
p.s. changes on line 61/63 were not supposed to be included, removed in c7d683e

closed 2018-08-03 by leonardt

David-Durst, 2018-08-03:
Confirmed, this solves the problem.