phanrahan/magma #238: Failing CoreIR Clock Wiring Pass On Magma Circuit

David-Durst, 2018-08-03:
When I run the test https://github.com/David-Durst/aetherling/blob/e7dc587694476612c455c139946adbed55c4dfcc/tests/test_native_line_buffer.py#L33-L55, I get the following error:

```Shell
Testing started at 12:51 AM ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pydev/pydevd.py" --multiproc --qt-support=auto --client 127.0.0.1 --port 59828 --file "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --target test_native_line_buffer.py::test_multiple_sipo -- -s
Connected to pydev debugger (build 182.3684.100)
pydev debugger: process 31233 is connecting

Launching pytest with arguments -s test_native_line_buffer.py::test_multiple_sipo in /Users/durst/dev/W17-8/aetherling/tests

============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 1 item
test_native_line_buffer.py In Run Generators
Done running generators

Assertion failed: (0), function die, file context.cpp, line 112.
Running pass rungenerators
In Run Generators
Done running generators
Running pass wireclocks-coreir
ERROR: Cannot connect multiple outputs to an inputs
In Module: mapParallel
  op_0.CLK : coreir.clkIn <== self.CLK[0]
  op_0.CLK : coreir.clkIn <== self.clk

I AM DYING!

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

Why are there multiple clocks being wired to the same thing? Is there a magma clock wiring pass competing with CoreIR's clock wiring pass?

leonardt, 2018-08-03:
I suspect this issue is localized to the coreir-wireclocks pass. If I make the following change (takeout the wireclocks-coreir call)

```diff
diff --git a/tests/test_native_line_buffer.py b/tests/test_native_line_buffer.py
index b409b69..92dc1d0 100644
--- a/tests/test_native_line_buffer.py
+++ b/tests/test_native_line_buffer.py
@@ -46,7 +46,7 @@ def test_multiple_sipo():
     mod = GetCoreIRModule(cirb, testcircuit)
-    for p in ["rungenerators", "wireclocks-coreir", "verifyconnectivity-noclkrst",
+    for p in ["rungenerators", "verifyconnectivity-noclkrst",
                              "flattentypes", "flatten", "verifyconnectivity-noclkrst", "deletedeadinstances"]:
         print("Running pass {}".format(p))
         c.run_passes([p], namespaces=["aetherlinglib", "commonlib", "mantle", "coreir", "global"])
```

I get
```
❯ pytest -s -x tests/test_native_line_buffer.py -k test_multiple_sipo
============================================================================================= test session starts =============================================================================================
platform darwin -- Python 3.6.5, pytest-3.2.3, py-1.4.34, pluggy-0.4.0
rootdir: /Users/lenny/google_drive/repos/aetherling, inifile:
plugins: xdist-1.22.2, forked-0.2, cov-2.5.1, rerunfailures-4.0, pep8-1.0.6
collected 7 items
tests/test_native_line_buffer.py In Run Generators
Done running generators
Running pass rungenerators
In Run Generators
Done running generators
Running pass verifyconnectivity-noclkrst
  Def:
    Instances:
      const_1_1 : const(width:1)
      inst0 : mapParallel(numInputs:1, operator:global.SIPO4CE)
    Connections:
      const_1_1.out <=> inst0.CE
      inst0.I[0] <=> self.I
      inst0.O[0] <=> self.O
ERROR: {multiple_sipo_test}.inst0.CLK Is not connected
I AM DYING!
Assertion failed: (0), function die, file context.cpp, line 112.
zsh: abort      pytest -s -x tests/test_native_line_buffer.py -k test_multiple_sipo
```
So perhaps the wireclocks-coreir pass is trying to wire up the clock more than once?

leonardt, 2018-08-03:
Actually scratch that last note, it looks like it's complaining for a different clock that isn't wired, so it could still be an issue with multiple entities trying to wire up a clock

leonardt, 2018-08-03:
The wireclocks-coreir pass checks if the clock is already and connected and skips if so, so I don't think that's the issue, see https://github.com/rdaly525/coreir/blob/4ea12ff9f1e6b414306d8cc5b9024d13cd0db18b/src/passes/transform/wireclocks.cpp#L38

It seems like this circuit has a nested clock? (based on the `self.CLK[0]`) I suspect that's the issue

David-Durst, 2018-08-03:
Yes, I had problems with that in magma as well. Am I not allowed to have nested clocks? Map is just lifting the ports on the modules inside of it to its own interface. Should I make a special case for those called CLK?

David-Durst, 2018-08-03:
Adding a special case for map and reduce for clock would add some complexity to Aetherling's code.

David-Durst, 2018-08-03:
I wrote the map and reduc in CoreIR. I can just not expose any port name clk on the map. How does coreir figure out what is a clock port?

leonardt, 2018-08-03:
It checks for the clock namedType, I will look at wireclocks-coreir, we may be able to make it support nested clock types

David-Durst, 2018-08-04:
https://github.com/rdaly525/coreir/pull/591 - this makes map not lift the clock ports. My unit test passes now. However, I still think wireclocks-coreir should support nested clocks.

leonardt, 2018-08-04:
One question, how should wireclocks know which clock to wire up to which instance? It seems like this will be difficult to do automatically. You could provide a new type for each instance/clock pair, so it could match based on the types. In this case, it might make sense to have your definition handle the wiring manually since it knows the mapping?

The default clock wiring was meant to serve for simpler cases (e.g. one top-level clock port wiring to instances with one unwired clock). If we can come up with a reasonable mapping for this case, we can do it, but right now it seems like if you have an array of clocks and a set of instances in the definition with unwired clocks, there's no way to know which is supposed to go to which.

David-Durst, 2018-08-04:
All the instances should get the same clock. There are an array of clock ports as that is the standard behavior of map: make an array of each port.

I'm not sure I follow why this case is different from the normal case. In a normal circuit, there are lots of unwired clock ports and by default they all get the same clock. In the nested clock ports case, there are also lots of unwired clock ports and by default they should should all get the same clock.