phanrahan/magma #239: Support For Multidimensional Indexing Into Nested Arrays

David-Durst, 2018-08-05:
Let modDef be a module with:
1. in port I of type: Array(4, Array(3, BitIn))
2. out port O of type: Array(3, Array(4, BitOut))

I would like to be able do  to do the following:
```python
for i in range(3):
   wire(in.I[:][i], out.O[i])
```

An alternative solution would be to allow reordering multidimensional arrays so that I can just wire up the entire arrays after reordering. However, multidimensional indexing seems like a more flexible solution that could be helpful in other situations.

leonardt, 2018-08-07:
Definitely seems like a good feature, also related to https://github.com/phanrahan/magma/issues/165

leonardt, 2020-08-07:
Added by #777

closed 2020-08-07 by leonardt