phanrahan/magma #248: 0 length Array only complains downstream

rsetaluri, 2018-08-13:
We can create 0 length arrays in Magma: `T = magma.Array(0, ...)` and even use them in the course of wiring. Do we want to allow this? The core ir backend doesn't support this...unsure if the verilog backend does.

leonardt, 2018-08-13:
I'd vote for not allowing this, seems like a weird case that will just introduce complexity in user and internal code.

rdaly525, 2018-08-14:
I also vote for disallowing this, but I should note that chisel /firrtl guys found a useful case for 0 length wires in generating address bit widths for memories.

phanrahan, 2018-08-14:
Makes sense to not allow that, although I do remember the chisel use case. Let's think about it a bit more before making the change.

leonardt, 2018-08-14:
Can you expand on the address width for memory example? That might help us guide a way to include the feature but in a special way. If it's useful, we could add some explicit version of it (maybe special type or something), so we can support it, but still guard against cases where the user accidentally creates a length 0 wire.