phanrahan/magma #254: Add options to generate verilog as separate files per module or modules organized by original source file

leonardt, 2018-08-22:


rsetaluri, 2018-08-27:
I think this is an important feature for generating good verilog. One question is whether we need to change coreir or not. I see a few possibilities:

1. For each module (or set of modules), generate separate `<name>.json` files. Then just call `coreir -i <name>.json -o <name>.v` for each module. Requires only a magma change.
2. For each module (or set of modules), generate a single `<top>.json` file as normal. Then send an option to coreir that would split the files, ala: `coreir -i <top.json> -o <prefix> --split-verilog-files`
3. For each module (or set of modules), generate separate `<name>.json` files. Then call `coreir -i <name1>.json <name2>.json ... -o <prefix> --split-verilog-files`. This way there is no ambiguity regarding colliding namespaces or module names etc.

(2) is the most "correct" way to do it, but (1) is the easiest and least invasive.

@rdaly525 @leonardt thoughts?

rsetaluri, 2018-08-27:
After a brief conversation with Lenny, I think we agree that (2) is the best option. We should try to add a flag in coreir to split into verilog files according to some schemes, and also parse in something like `-o <prefix>` and figure out that the output is verilog from the `--split-verilog-files` option.

rdaly525, 2018-08-29:
One minor concern for 2) is how to automatically determine the name of each file. Currently you have to explicitly pass the name of the verilog file to the coreir command

rsetaluri, 2018-08-29:
I'm working on an interface for that (tbh think the interface to the coreir binary could be cleaned up). What I was thinking was requiring something like:

`coreir -i in.json -o ouptut_dir/*.v --split`

That way you can use the same logic to infer that you want to output to verilog, and use the prefix to know which directory to place the files. As a first pass, we can use the module name to determine the filename (we can implement something more complex such as mapping to original filename in magma or use some other metadata). This is the most flexible & least invasive thing that will work. If `--split` is included we can assert that the `-o` argument is of the form `some/dir/path/*.v`.

rdaly525, 2018-08-29:
@rsetaluri, that seems like a reasonable interface. If you implement it, can you still make the split generic to the other file formats like json? Or at the very least detect an assert a Not Yet Implemented error.

rsetaluri, 2018-08-29:
Ok, I'll make sure to do that.

rsetaluri, 2018-09-06:
Resolving as https://github.com/rdaly525/coreir/pull/612 is merged.

closed 2018-09-06 by rsetaluri