phanrahan/magma #259: Feature request: Specify which ports in the IO list are logical "inputs" and "outputs"

rdaly525, 2018-08-23:
This would allow you to do the nice instance apply operator to more complex types like ready/valid.

phanrahan, 2018-08-28:
I agree. We overload input with In, and output with Out.

I used to allow ["input I", In(Bit)]. That might be a way to do this. Other suggestions welcome.

rdaly525, 2018-08-29:
I think for more complex types, using In and Out is not viable (for example if I have an upstream ready/valid which I want to specify as an "input", and a downstream ready/valid I want to specify as an "output")

Since this only applies to circuit interfaces, it would make sense to set another member or use some sort of decorator.

For example something along the lines of:

```
class Foo(Circuit):
  name = "Foo"
  IO = ["upstream", Flip(ReadyValid(Bits(16))), "downstream", ReadyValid(Bits(16)), "otherports", Tuple(...)]
  apply_overload = dict(input:"upstream",output:"downstream")
  def definition(io):
     ...

```
The behavior of this would be that you could call Foo with the readValid bundle that connects to upstream, and would return the readyValid bundle for downstream. It would also require you to manually wire up the "otherports" separate from calling the instance.

phanrahan, 2018-08-30:
That method would also work. They both designate certain arguments as inputs or outputs.

I propose the following which should work already. Mixed mode types are not considered inputs or outputs for now. You need to explicitly wire them up. For example, `wire( producer.downstream, consumer.upstream)`. We also have a feature request to allow wiring up arguments using their names. For example, consumer(downstream=producer.upstream)`.

rdaly525, 2018-08-30:
For now I think that is fine, but in the future mixed mode types would be nice to consider inputs/outputs from the instance application perspective.

For example I think for UART it was requested that it would be cool to write something along the lines of:

`wire(io.tx,transmitter(fifo(receiver(io.rx))))
`
opposed to:
```
wire(io.rx, receiever.in)
wire(receiver.out,fifo.in)
wire(fifo.out,transmitter.in)
wire(transmitter.out,io.tx)
```