phanrahan/magma #263: Don't add to a circuit when using DefineCircuit

rsetaluri, 2018-08-24:
There is an important difference in semantics between subclassing `Circuit` and using `DefineCircuit` -- when we subclass, if the name hasn't changed, the body of `definition` will not execute. However, with `DefineCircuit` the body will re-execute, adding instances. There should be parity b/w these 2 modes. And ideally caching should not affect the functionality just the performance (though this may be hard since we are metaprogramming and using names as cache keys).

With `DefineCircuit`, we should at least issue a warning if a circuit w/ the same name has been "defined" twice. Ideally we would ignore the body as we do with subclassing.

phanrahan, 2018-08-28:
Yes, good point. Nasty bug. This caching stuff is really a pain.

leonardt, 2020-08-07:
I think we can close this since we're dropping DefineCircuit support

closed 2020-08-07 by leonardt