phanrahan/magma #265: Explicitly warn user when trying to use circuit def ports rather than instance ports

rsetaluri, 2018-08-24:
The code:

```python
bar = magma.DefineCircuit("bar", "I", magma.In(magma.Bits(2)), "O", magma.Out(magma.Bits(2)))
magma.wire(bar.I, bar.O)
magma.EndCircuit()

foo = magma.DefineCircuit("foo", "I", magma.In(magma.Bits(2)), "O", magma.Out(magma.Bits(2)))
bar_inst = bar()
magma.wire(foo.I, bar.I)
magma.wire(bar.O, foo.O)
magma.EndCircuit()
```

Reults in the error:

```
magma:WARNING:tmp.py:11: Adding the output `foo.I[0]` to the wire `bar.I[0]` which already has output(s) `[bar.I[0]]`
magma:ERROR:    magma.wire(foo.I, bar.I)
```

Instead we should say something like: `using cirucit def "foo" as instance"

leonardt, 2018-08-24:
It seems like we can make is such that `bar.I` is only wireable within the context of it's definition. So, if we're inside a `definition` method, or `EndCircuit` hasn't been called for the circuit, otherwise we can mark it as unwireable and raise the error.

rsetaluri, 2018-08-24:
Makes sense. I agree.

phanrahan, 2018-08-28:
@leonardt making it wirable only within the context of a definition seems good to me.