phanrahan/magma #268: Save generator parameters in generated modules

leonardt, 2018-08-29:
There's already a decorator that does this, see . https://github.com/phanrahan/magma/blob/master/magma/circuit.py#L564 but we need to standardize it's use by people writing generators (e.g. in mantle).

rsetaluri, 2018-08-29:
We could also add a hook to auto generate the name based on ^.

rsetaluri, 2018-09-08:
What about something like

```python
def save_parameters(fn):

    signature = inspect.signature(fn)
    names = list(signature.parameters.keys())[1:]

    def _params(self):
        return {name: getattr(self, name) for name in names}

    def _fn(this, *args, **kwargs):
        if not hasattr(type(this), "params"):
            setattr(type(this), "params", _params)
        return fn(this, *args, **kwargs)

    return _fn

class Foo(Generator):
    @save_parameters
    def __init__(self, width):
        super().__init__()

        self.width = width

        self.add_ports(
            I=magma.In(magma.Bits(width)),
            O=magma.Out(magma.Bits(width)),
        )

        self.wire(self.I, self.O)

    def name(self):
        return "Foo"

foo = Foo(16)
print (foo.params())
print (foo.params()["width"])
print (foo.width)
magma.compile("foo", foo.circuit(), output="coreir")
```

This assumes that all the arguments to the constructor are saved as members and span all the generator arguments. Of course this does not cover more complicated cases, or derived parameters. But I think it's a reasonable convention.

leonardt, 2020-08-07:
@rsetaluri does the new Generator2 pattern save a references to the generator arguments in the generated Circuit? If not, it should be pretty simple to add.

rsetaluri, 2020-08-07:
Good point. It doesn't right now but leave this open for tracking adding that - should be very easy.