phanrahan/magma #269: [feature request] Automatic uniquification

leonardt, 2018-08-29:
This presents an alternative to cacheing that was implemented by Genesis2. This is going to change some fundamental semantics about our generator system, so it would be good to review the design of this and understand the implications. The feature request here is to automatically determine if a newly generated circuit is identical to a previously defined circuit. If so, use the previously defined one, otherwise, generate a new unique name for the circuit.

One mechanism we could use to do this is the fact that we already store circuits in a name based cache. If we are defining a new circuit with a name in the cache, we would need some mechanism for comparing the cached version of the circuit to the newly defined one, and then determining if we can uniquify them (return the cached version so there's only one copy of the same definition), or if we need to generate a new circuit with a unique name.

It seems like this strategy eschews name based uniqueness for a more thorough check, but it's not clear (based on Ofer's thesis https://stacks.stanford.edu/file/druid:wv793rg3775/os_thesis_final-augmented.pdf) how uniqueness is determined. I suspect it may be string comparing the generated verilog, which would work for the genesis system. Could we do something like string comparing the repr in magma?

More generally, I think if we decide to implement this feature, we'd be moving away from our name based caching mechanism, which is something that I think we have experience pains with. Instead, we would allow generators to re-generate copies of circuits, and have the magma internals deal with determining whether a new, unique circuit needs to be defined (with a unique name), or if it can refer to an existing definition.

rdaly525, 2018-08-29:
I want to note that the way that CoreIR handles this is caching explicitly based on the generator parameters. That being said all generator parameters in CoreIR are serializable, so easily comparable.

It seems that regardless of how you do the caching, the ability to generate a name is a very good feature to have. 

Also important to note is that false positives (different generator parameters falsely aliasing in the cache) is *really* bad, but false negatives (similar generator parameters that would generate the same structural circuit not aliasing in the cache) do not matter for anything other than a performance perspective.

leonardt, 2018-08-29:
In python we can't automatically cache based on generator parameters. We've already ran into this with a generator that accepts a list as a parameter, which is unhashable in Python.

rsetaluri, 2018-08-29:
It is a good point about false positives and false negative. But I'm not sure that any automatic way *based on generator parameters* can guarantee no false negatives. Perhaps this is not that common, but because python is such a dynamic environment (e.g. simple things such as global variables), we may not be able to say that a circuit is purely a function of the args to the generator function.

I advocate to using generator classes vs. functions moving forward (we should probably keep mantle as is, i.e. keep using functions there). I think that gives us more structure to allow us to solve this problem. And each generator class can explicitly declare it's parameters. (or use decorators etc. to make this very easy to do).

phanrahan, 2018-09-06:
I reread Ofer's thesis and looked at the code. He uniquifies based on comparing the output verilog.

One issue here is that if we are generating coreir, we may not even have verilog. Also, as we go down the generator linking path, we have no guarantees that the generators called are pure functions of the parameters. We need to sit down and hash through all the scenarios.

I do vote for going with class-based generators. And for developing a robust way to compare parameters.

rsetaluri, 2018-09-07:
Agreed. I think the main thing that we have to hash out is what kinds of dynamism are allowed on these classes and what services we provide vs what is automated (e.g. unique naming, generating circuits).

rdaly525, 2018-09-07:
Identical CoreIR graphs should produce identical json, so we could diff on that.

phanrahan, 2018-09-07:
I discussed this issue with Ofer.

1. He initially determined uniqueness using names and parameters, but he found that some generators were not pure functions, and so that method was not robust. An example of this is when instances inspect parameters in other instances, or the global configuration sets different parameter values for different instances in the hierarchy.

2. He confirmed that Genesis2 compared verilog files, but he agrees this is not the idea method. He would prefer a method based in comparing the graph of circuit instances.