phanrahan/magma #270: [feature request] Generating individual files

leonardt, 2018-08-29:
The generation of individual verilog files can be handled at the coreir level, unless someone explicitly requires generating multiple coreir files.

There's also a request for generating file lists, with options for generating DV only lists (what is DV, design verification?), synthesis only list, and a merged list.

We should clarify the exact requirements of this feature:
* What is contained in these generated file lists? 
* How do we specify the "type" of modules (e.g. DV or synthesis)?

rsetaluri, 2018-08-29:
duplicate of #254 ? Also see https://github.com/rdaly525/coreir/issues/609, assigned to me.

leonardt, 2018-08-29:
Related, but also adds the generating file lists. I'll merge that into the existing issue once we get clarification on what that actually means.

rsetaluri, 2018-09-06:
Note that #254 is closed. There still maybe more complex things to do such as mapping source magma filenames with output filenames, or including a list of generated files. Leaving this open primarily for the list-of-files request.

leonardt, 2018-09-10:
Renamed title to match the outstanding issue (file listings for multiple file output)