phanrahan/magma #277: Add mechanism to convert bits to clocks when parsing verilog

rsetaluri, 2018-09-06:
CoreIR type checks clock wiring; it does not allow wiring Bit to Clock. When parsing verilog everything is Bit/s. We either need to add an option `clocks=[]` to Declare/DefineFromVerilog which dictates that certain port names should be Clock not Bit (with appropriate checks that this conversion makes sense). Or we could add a lifting pass that allows conversion from Bit to Clock in general.

rdaly525, 2018-09-06:
Presumably when you are wiring to the lifted verilog, you know exactly where the clock is. Why not just explicitly convert from the clock to a bit while wiring?

rsetaluri, 2018-09-07:
That is an option that will not error, but is not identical to what I’m suggesting. Primarily because we get the extra typing of the wire being a clock (then we could even do the automatic clock wiring pass). Maybe those benefits are not worth it, but it seems like the right thing to do is convert the Bit to Clock, not the other way around.

rdaly525, 2018-09-07:
Yeah, it sounds like a good solution is then to pass in what are clock ports to the Declare/DefineFromVerilog in magma, then when you construct the wrapper coreir module, just use a clock port(s) in the type.

closed 2018-09-07 by rsetaluri