phanrahan/magma #282: [feature request] Make it easier to define type aliases and composite types

leonardt, 2018-09-10:
See https://github.com/phanrahan/magma/blob/09d08ce35a3d95006d59e3f698775c1081069e20/tests/test_type/test_new_types.py#L7 for some examples.

I think there are some ways to improve the user experience for defining new types.

Cases:
1. Type aliasing, e.g. `typedef Coordinate m.Bits`. In this case, we just want a new name (possibly nominal type matching?) with the same behavior as Bits. The one detail here is that `Bits` is a type constructor, so we need to add some mechanism for this to workout. Possible syntax `Coordinate = m.type_alias('Coordinate', m.Bits)` or `Coordinate = `m.Bits.alias()`. This probably covers quite a bit of custom types that extend Bits or Array.
2. Composite types, e.g.
    ```
    def Point2D(num_bits):
        return m.Tuple(x=Coordinate(num_bits), y=Coordinate(num_bits))
    ```
    In this case, we need to define a new type constructor, which is fine. However, now the user also needs to implemented Point2DType and Point2DKind to get the full suite of Type related features, e.g. `isinstance(port, Point2DType)`. I think there's motivation to clean up this part of the code. Basically, Caleb and I have discussed possible ways to rearchitect the type system code such that one could say `isinstance(port, Point2D)` (basically changing type constructors to be classes which override the isinstance behavior). Also, defining a new type constructor should have an option to generate the `Type` and `Kind` class definitions automatically, so long as the user doesn't want to extend it. So, if you're just defining a simple composite type that preserves all of Tuple semantics, then there should be no need to actually write a `Type` and `Kind` object. Moving to the class pattern for type constructors might resolve this issue (since then the Type and Kind objects may just be used internally and the user won't have to worry about it). Finally, there's the case that the user may want to extend the types to define new functionality. I think this is the case where they will need to define their own `Type` class (and possible `Kind` class?). Again, I think there might be ways to improve this from a user experience perspective (perhaps integrating into the type constructor).

In summary, I think defining custom types, particularly type aliases and composite types, is an essential language feature so we'll want to improve the user experience for doing this.

leonardt, 2018-09-10:
One issue about type aliases: do we do nominal type checking? For example, Rust does not (see https://doc.rust-lang.org/book/2018-edition/ch19-04-advanced-types.html#creating-type-synonyms-with-type-aliases), so type aliasing is merely a convenience. Here's a direct quote, note they use synonym instead of alias
> The main use case for type synonyms is to reduce repetition

The main issue here is, if I say (pseudo-syntax)
```
NewType = m.alias(m.Bits)
```

then, does the following code type check?

```
a = NewType(2)
b = m.Bits(2)
m.wire(a, b)
```

leonardt, 2018-09-10:
It seems like one viable path is to introduce type aliasing as purely a convenience mechanism, and require explicitly introducing a new type (through some other mechanism) if you want stronger type checking

leonardt, 2018-09-11:
```
Coordinate = m.Bits 
```

and

```
Coordinate2 = m.Bits(2)
```

work fine unless you want the `repr`/`str` to show the aliased name. Right now it shows

```
>>> import magma as m
>>> Coordinate = m.Bits
>>> Coordinate(3)
Bits(3)
>>> Coordinate2 = m.Bits(2)
>>> Coordinate2
Bits(2)
```

This is okay for structural typing for aliases, but it could be confusing for debugging. I'm not sure of a way to associate the name for debugging without an explicit function call.

phanrahan, 2018-09-11:
i get it now. You are right, the name will not change.

So, an type alias should just change its name.

`NewType` creates a type that type checks differently.

leonardt, 2018-09-11:
Yes agreed, that makes sense to me.