phanrahan/magma #285: Unsigned Division Operator Doesn't Work

David-Durst, 2018-09-12:
I get the following output from running https://github.com/David-Durst/aetherling/blob/45437e120f79a606cdf4ea2d042cfb7c3f7584b7/tests/test_delayed_buffer.py#L10:

```
Launching pytest with arguments -s test_delayed_buffer.py::test_delayed_buffer in /Users/durst/dev/W17-8/aetherling/tests

============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
plugins: xdist-1.22.5, forked-0.2
collected 1 item
test_delayed_buffer.py In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
Starting topological sort
topo_order.size() = 38
numVertices(g)    = 38

Unsupported node: inst7 has operation name: coreir.udivAssertion failed: (false), function updateNodeValues, file interpret.cpp, line 1040.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

leonardt, 2018-09-12:
This looks like an issue in the coreir simulator

@dillonhuff can you interpret this error message?

leonardt, 2018-09-12:
My interpretation of it says that `coreir.udiv` is not supported in the interpreter

phanrahan, 2018-09-12:
I added support for promoting ints when wiring. But I don't think we have added support for promoting ints to the operators. @leonardt did you do that?

dillonhuff, 2018-09-13:
@leonardt @David-Durst I just added udiv in this pull request:

https://github.com/rdaly525/coreir/pull/615

dillonhuff, 2018-09-24:
Closing this now that the division pull request has been merged

closed 2018-09-24 by dillonhuff