phanrahan/magma #300: Issue with overwriting output to "coreir-verilog"

rsetaluri, 2018-09-26:
The new change that changes the backend from "verilog" to "coreir-verilog" automatically when mantle is in use with target "coreir" is causing issues (see https://travis-ci.com/StanfordAHA/garnet/builds/85948541). In particular here, wrapped verilog is getting compiled through coreir, though this does not work as of now. Unfortunately, though this particular circuit does not use mantle *at all*, because we are running all tests (some of which *do* use mantle) at once through pytest, magma says that mantle is being imported, and therefore does the transformation of the backend.

Ideally, `magma.compile()` checks that the circuit being compiled itself contains any mantle components, rather than checking simply that mantle was at some point imported. This may be hard or not possible with current hooks, so I propose that at very least we have some way of turning off this overriding mechanism (or alternatively, make it "opt in").

leonardt, 2018-09-26:
Okay, I was worried something like this would happen.

The reason for this change was to avoid the issue where people got garbage verilog output or errors when trying to compile with the verilog output and the mantle target is using coreir.  I'm going to revert the change for now and instead just log a warning if you try to do the above (instead of automatically changing to verilog), so we'll revert to the old behavior.

leonardt, 2018-09-26:
This change reverts to the old behavior https://github.com/phanrahan/magma/commit/654629715ea3c30ed7ee945fdbc28a9a556c2f39 except it now prints out a warning when we encounter the case of using `output="verilog"` and `mantle_target="coreir"`.

leonardt, 2018-09-26:
If I understand correctly, the above issue could be remedied by having coreir accept "wrapped verilog" for module definitions. Then, all the garnet code could be changed to use the coreir backend.

Also, another improvement to avoid people compiling with the wrong output is to have the magma compile default output be coreir instead of verilog. This way, the default is consistent with the default mantle target (coreir). I think the major blocker for this would be compiling wrapped verilog.

leonardt, 2018-09-26:
Just pushed a test run here https://github.com/StanfordAHA/garnet/pull/119/commits/0f6229e1d843c57c4a76641e00a90c6de8dc5535

Also submitted a PR with a mocked magma test. There's also a mantle test here: https://github.com/phanrahan/mantle/blob/master/tests/test_coreir/test_compile_coreir_verilog.py passing locally for me.

closed 2018-09-26 by leonardt

rsetaluri, 2018-09-26:
Awesome, thanks @leonardt