phanrahan/magma #302: [feature proposal] Convenience mechanism for construct-in circuit definitions from higher order circuit combinators

leonardt, 2018-10-03:
Right now, if we want to use higher order combinators to construct definitions, we need to do them inside a circuit definition, see the register definition for an example: https://github.com/phanrahan/mantle/blob/master/mantle/common/register.py#L66-L70

The issue here is that the `IO` of the instance returned from `join` is exactly equal to the IO of the enclosing definition.  So really, the convenience of using the higher order combinator is lost when I have to still specify the IO explicitly for the enclosing definition. Furthermore, we have to perform the wiring from the instance to the IO.

It would be convenient to have some mechanism for lifting the result of a higher order circuit into a definition (rather than an instance) or have some other path for creating a definition out of the operators rather than an instance.

phanrahan, 2018-10-05:
One way to think of this is that we would convert an anonymous circuit to a definition. I agree that would be nice to have.

leonardt, 2018-10-05:
Yes, I think that's exactly what we want. In fact, it would perhaps be a better name to use `AnonymousInstance` for what an `AnonymousCircuit` currently is, and have `AnonymousCircuit` be the corresponding definition.