phanrahan/magma #304: Statically elaborated for loops inside circuit combinational

leonardt, 2018-10-12:
Feature request from @priyanka-raina

This would make it more convenient for doing array logic where the indexing logic is dependent on computations.

I think one strategy for implementing this would be a metaprogramming system ala Terra, where we can use Python to metaprogram `@circuit.combinational` programs.

Alternatively, we could just slap on a statically unrollable for loop construct, which would be faster time to solution versus a more general metaprogramming framework.

@priyanka-raina if you have a chance, could you provide a simple example of some array indexing logic that's hard to do with maps and reduces?

It could be a snippet of bluespec code, or even better, some Python/magma code that you wish you could write.

priyanka-raina, 2018-10-17:
For example, what is the best way to write this bluespec function using magma combinational?
```
function Tuple3#(box2d_t, polygon_t, colors_t) bounding_box (
    polygon_t poly, 
    colors_t c, 
    point2d_t screen_max, 
    SampleSize sample_size
  );
   
    box2d_t box = vector(vector(0, 0), vector(0, 0));
    
    for (Integer j = 0; j < 2; j = j + 1) begin
      Vector#(vertices, Coordinate#(n)) v;
      for (Integer i = 0; i < valueOf(vertices); i = i + 1) v[i] = poly[i][j];
      box[0][j] = fold(min, v);
      box[1][j] = fold(max, v);
    end
    
    Bit#(fractional_bits) msaa_fractional_mask = 
      sample_size == ONE_PIXEL     ? 0 : 
      sample_size == HALF_PIXEL    ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) :
      sample_size == QUARTER_PIXEL ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 2) :
      sample_size == EIGHTH_PIXEL  ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 2) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 3) : 
                                    0;
    
    Bit#(integer_bits) msaa_integer_mask = '1;

    for (Integer j = 0; j < 2; j = j + 1)
      for (Integer i = 0; i < 2; i = i + 1)
        box[i][j] = unpack(pack(box[i][j]) & {msaa_integer_mask, msaa_fractional_mask});
     
    for (Integer i = 0; i < 2; i = i + 1) begin
      box[0][i] = box[0][i] < 0 ? 0 : box[0][i];
      box[1][i] = box[1][i] > screen_max[i] ? screen_max[i] : box[1][i];
    end
  
    return tuple3(box, poly, c);
  endfunction
```

priyanka-raina, 2018-10-17:
Adding the full module for context:
```
import Vector::*;
import RegFile::*;
import VectorOps::*;
import Trace::*;
import MiscFunctions::*;

typedef enum {
  ONE_PIXEL = 4'b1000,
  HALF_PIXEL = 4'b0100,
  QUARTER_PIXEL = 4'b0010,
  EIGHTH_PIXEL = 4'b0001
} SampleSize deriving (Bits, Eq);

typedef Int#(n) Coordinate#(numeric type n);
typedef Vector#(axes, Coordinate#(n)) Point#(numeric type axes, numeric type n);
typedef Vector#(vertices, Point#(axes, n)) Polygon#(numeric type vertices, numeric type axes, numeric type n);

typedef UInt#(n) Color#(numeric type n);
typedef Vector#(color_channels, Color#(n)) Colors#(numeric type color_channels, numeric type n);

typedef Vector#(2, Coordinate#(n)) Point2D#(numeric type n);
typedef Vector#(2, Point2D#(n)) Box2D#(numeric type n);
    
interface ComputeBoundingBox#(
  numeric type i, // integer bits in fixed point number
  numeric type f, // fractional bits in fixed point number
  numeric type vertices,
  numeric type axes, 
  numeric type color_channels,
  numeric type pipe_depth);

  method ActionValue#(Maybe#(Tuple3#(Box2D#(TAdd#(i, f)), Polygon#(vertices, axes, TAdd#(i, f)), Colors#(color_channels, TAdd#(i, f))))) get_bounding_box (Maybe#(Tuple4#(Polygon#(vertices, axes, TAdd#(i, f)), Colors#(color_channels, TAdd#(i, f)), Point2D#(TAdd#(i, f)), SampleSize)) in);

endinterface

module mkComputeBoundingBox(ComputeBoundingBox#(
    integer_bits, 
    fractional_bits,
    vertices, 
    axes, 
    color_channels, 
    pipe_depth
  )) provisos (
    NumAlias#(n, TAdd#(integer_bits, fractional_bits)),
    Alias#(polygon_t, Polygon#(vertices, axes, n)),
    Alias#(colors_t, Colors#(color_channels, n)),
    Alias#(point2d_t, Point2D#(n)),
    Alias#(box2d_t, Box2D#(n)),
    Add#(1, a__, vertices)
  );
  
  Reg#(Maybe#(Tuple3#(box2d_t, polygon_t, colors_t))) pipe_r <- mkReg(Invalid);
 
  function Tuple3#(box2d_t, polygon_t, colors_t) bounding_box (
    polygon_t poly, 
    colors_t c, 
    point2d_t screen_max, 
    SampleSize sample_size
  );
   
    box2d_t box = vector(vector(0, 0), vector(0, 0));
    
    for (Integer j = 0; j < 2; j = j + 1) begin
      Vector#(vertices, Coordinate#(n)) v;
      for (Integer i = 0; i < valueOf(vertices); i = i + 1) v[i] = poly[i][j];
      box[0][j] = fold(min, v);
      box[1][j] = fold(max, v);
    end
    
    Bit#(fractional_bits) msaa_fractional_mask = 
      sample_size == ONE_PIXEL     ? 0 : 
      sample_size == HALF_PIXEL    ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) :
      sample_size == QUARTER_PIXEL ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 2) :
      sample_size == EIGHTH_PIXEL  ? 1 << (fromInteger(valueOf(fractional_bits)) - 1) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 2) | 
                                    1 << (fromInteger(valueOf(fractional_bits)) - 3) : 
                                    0;
    
    Bit#(integer_bits) msaa_integer_mask = '1;

    for (Integer j = 0; j < 2; j = j + 1)
      for (Integer i = 0; i < 2; i = i + 1)
        box[i][j] = unpack(pack(box[i][j]) & {msaa_integer_mask, msaa_fractional_mask});
     
    for (Integer i = 0; i < 2; i = i + 1) begin
      box[0][i] = box[0][i] < 0 ? 0 : box[0][i];
      box[1][i] = box[1][i] > screen_max[i] ? screen_max[i] : box[1][i];
    end
  
    return tuple3(box, poly, c);
  endfunction

  method ActionValue#(Maybe#(Tuple3#(box2d_t, polygon_t, colors_t))) get_bounding_box (
    Maybe#(Tuple4#(polygon_t, colors_t, point2d_t, SampleSize)) in);
    
    let out = tagged Invalid;
    if (isValid(in)) begin 
      let {poly, color, screen_max, sample_size} = validValue(in);
      out = tagged Valid bounding_box (poly, color, screen_max, sample_size);
    end
    pipe_r <= out;
    return out; 
  endmethod

endmodule
```

leonardt, 2020-08-07:
Added via ast_tools

closed 2020-08-07 by leonardt