phanrahan/magma #319: Make it easier compile multiple times in the same program

rsetaluri, 2018-11-16:
Right now calling `magma.compile()` more than once in the same program is not guaranteed to always work. For example:

```python
import magma as m
import mantle

foo = m.DefineCircuit("foo", "I", m.In(m.Bit), "O", m.Out(m.Bit))
m.wire(~foo.I, foo.O)
m.EndDefine()

m.compile("foo", foo, output="coreir")

bar = m.DefineCircuit("bar", "I", m.In(m.Bit), "O", m.Out(m.Bit))
m.wire(bar.I & bar.I, bar.O)
m.EndDefine()

m.compile("bar", bar, output="coreir")
```
produces:

foo.json:
```json
{"top":"global.foo",
"namespaces":{
  "global":{
    "modules":{
      "foo":{
        "type":["Record",[
          ["I","BitIn"],
          ["O","Bit"]
        ]],
        "instances":{
          "inst0":{
            "modref":"corebit.not"
          }
        },
        "connections":[
          ["self.I","inst0.in"],
          ["self.O","inst0.out"]
        ]
      }
    }
  }
}
}
```

bar.json:
```json
{"top":"global.bar",
"namespaces":{
  "global":{
    "modules":{
      "foo":{
        "type":["Record",[
          ["I","BitIn"],
          ["O","Bit"]
        ]],
        "instances":{
          "inst0":{
            "modref":"corebit.not"
          }
        },
        "connections":[
          ["self.I","inst0.in"],
          ["self.O","inst0.out"]
        ]
      }
    }
  }
}
}
```

Notice the error in `bar.json`. Not sure exactly why this is happening, but I'm guessing it has to do with the reuse of the CoreIR context in the backend. I even tried using the `reset_context` hook, but this causes other issues (pycoreir complains that two compilations units have different contexts I think). It would be good to make this much easier at the user level.

leonardt, 2018-11-16:
Great thanks, this has been a known lurking issue and it's good to have a small example to reproduce it. I'll investigate it when I get a chance.

leonardt, 2019-01-25:
Finally figured this out, the issue is that the JSON pass (used to emit the file) is an analysis pass which will only run again if the graph is modified. However, this logic is only for passes, so it doesn't record when we modify the namespace via the API (in this case, it happens when we compile the new circuit definition).

This PR (https://github.com/rdaly525/coreir/pull/656) adds a `markdirty` pass which we can use to force coreir to rerun the JSON pass (and therefor produce the new circuit).

The last issue to resolve is where we call `markdirty`. Something safe would be to have `markdirty` called before or after every time `compile` is invoked. This might incur some overhead of analysis passes being rerun, but I think that might be safest. This solution also doesn't require the user to call `markdirty` either.

Any thoughts?

rsetaluri, 2019-01-25:
Does rdaly525/coreir#656 solve this fully? Assuming that pass is run before compiling to CoreIR multiple times?

rsetaluri, 2019-01-25:
Got it. I think it's reasonable to keep it as something the user is required to invoke manually for now. Closing.

closed 2019-01-25 by rsetaluri