phanrahan/magma #333: Issue with uniquifying circuits with mantle40 DefineSub

splhack, 2019-01-18:
This simple test code doesn't compile due to `KeyError: 'Could not find key: Invert2'` for ice40 board.

    import magma as m
    from mantle import *
    class TestSub(m.Circuit):
        IO = ['I', m.In(m.SInt(2)), 'O', m.Out(m.Bit)]

        @classmethod
        def definition(io):
            o = mux([m.bit(0), m.bit(1)], io.I - m.sint(0, 2) <= m.sint(0, 2))
            m.wire(o, io.O)

The error only happens with mantle40. So pure coreir and coreir-verilog work just fine.

    $ magma -o coreir -t TestSub testsub.py
    (no error)
    $ magma -o coreir-verilog -t TestSub testsub.py
    (no error)

With mantle40, it causes the error.

    $ magma -b tinyfpga -o coreir -t TestSub testsub.py
    magma:WARNING:Could not compile circuit: ''Could not find key: Invert2''. Uniquifying anyway.
    ...
        raise KeyError("Could not find key: {}".format(key))
    KeyError: 'Could not find key: Invert2'

The problem is generating `_Sub` class twice here https://github.com/phanrahan/mantle/blob/cf6c792a0dab0e6ba0245b04904bed8c36a0bdb1/mantle/lattice/mantle40/arith.py#L80-L99

One is for subtract by `io.I - m.sint(0, 2)`, the other one is for signed comparison `<= m.sint(0, 2)`.

- https://github.com/phanrahan/mantle/blob/cf6c792a0dab0e6ba0245b04904bed8c36a0bdb1/mantle/common/arith.py#L11
- https://github.com/phanrahan/mantle/blob/cf6c792a0dab0e6ba0245b04904bed8c36a0bdb1/mantle/lattice/mantle40/compare.py#L171

Both `_Sub` (`Sub2`) classes look identical, but they are not the same. They have different id and different hash values. And eventually goes to `BuildInstanceGraphPass` `self.graph[]` https://github.com/phanrahan/magma/blob/ce970a2a1930e224b865af6c040e2e80b94a96c3/magma/passes/passes.py#L70 and they will exist as different keys in the dictionary.

`tsort` doesn't work well in this case, one `Sub2` key has no edges. `Sub2` is populated before `Invert2`. This is the reason of `Could not find key: Invert2` error.

splhack, 2019-01-18:
Forgot to mention, I thought cache_definition wraps the method and returns cached `Sub2`, but no.
https://github.com/phanrahan/mantle/blob/cf6c792a0dab0e6ba0245b04904bed8c36a0bdb1/mantle/lattice/mantle40/arith.py#L80

leonardt, 2019-01-28:
Should be fixed with #336, please reopen the issue if that's not the case.

closed 2019-01-28 by leonardt

splhack, 2019-02-07:
master(bb1350918d3ac40fc8b01ac8d0b359360c21de8b) has the same issue here.
But https://github.com/phanrahan/magma/pull/341 fixes the issue.

leonardt, 2019-02-07:
@splhack thanks for the heads up, we noticed that as well.