phanrahan/magma #345: Could Not Find Key Bug

David-Durst, 2019-02-08:
This unit test worked with the following versions from September-October:
magma - beeaec6bf5a5729aa94c83f203c5cf762efb8db3
mantle - dd3f3c4c6333273d92d079d256a2e293a9a36f7d
loam - de5c496e8c4a1fa4d3220a38cc4c40d375bcdbe9
coreir - 2cdb879029157eff85d04b59c2f1107e25f6adba
pycoreir - 92a4102d94ed0adf275e0d83670236e90467f42c

I updated to the latest of everything and it now fails:
```
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
In Run Generators
Done running generators
Numargs=1
F
tests/test_native_line_buffer.py:305 (test_1D_int8_line_buffer_1_15_39_13_1)
def test_1D_int8_line_buffer_1_15_39_13_1():
>       impl_test_1D_line_buffer(Array(8, Bit), 1, 15, 39, 13, -1)

test_native_line_buffer.py:307: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
test_native_line_buffer.py:195: in impl_test_1D_line_buffer
    a_1D_line_buffer_test(cirb, scope, in_arrays, magma_type, pixels_per_clock, window_width, image_size, output_stride, origin)
test_native_line_buffer.py:345: in a_1D_line_buffer_test
    namespaces=["aetherlinglib", "commonlib", "mantle", "coreir", "global"])
../../magma/magma/simulator/coreir_simulator.py:131: in __init__
    coreir_.compile(circuit, coreir_filename, context=self.ctx)
../../magma/magma/backend/coreir_.py:450: in compile
    backend.compile(main)
../../magma/magma/backend/coreir_.py:424: in compile
    self.compile_dependencies(defn_or_declaration)
../../magma/magma/backend/coreir_.py:413: in compile_dependencies
    self.modules[key.name] = self.compile_definition(key)
../../magma/magma/backend/coreir_.py:312: in compile_definition
    self.compile_definition_to_module_definition(definition, module_definition)
../../magma/magma/backend/coreir_.py:266: in compile_definition_to_module_definition
    coreir_instance = self.compile_instance(instance, module_definition)
../../magma/magma/backend/coreir_.py:196: in compile_instance
    module = lib.modules[name]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <coreir.util.LazyDict object at 0x11ed9cb00>
key = 'Register_has_ce_True_has_reset_True_has_async_reset_False_type_Bits_n_4'

    def __getitem__(self, key):
        if not key in self:
>           raise KeyError("Could not find key: {}".format(key))
E           KeyError: 'Could not find key: Register_has_ce_True_has_reset_True_has_async_reset_False_type_Bits_n_4'

../../pycoreir/coreir/util.py:25: KeyError
```

leonardt, 2019-02-08:
@rsetaluri and I will look at this, definitely related to the uniquification changes

leonardt, 2019-02-08:
Are you using your own coreir context object(s) anywhere in the code? (that is, not the magma coreir context singleton)

David-Durst, 2019-02-09:
Do you mean:
https://github.com/David-Durst/aetherling/blob/a10a970b6cae33bf004b9f645f8741fd0dac74f0/tests/test_native_line_buffer.py#L24 

If so, yes. That is the backbone of my testing.

David-Durst, 2019-02-09:
#347 fixes this.

closed 2019-02-09 by David-Durst