phanrahan/magma #352: Support Access To Nested Wires In CoreIR/Magma Simulator

David-Durst, 2019-02-13:
https://github.com/David-Durst/aetherling/blob/sim_issues/tests/helper_test_readyvalid.py#L54

It would be incredibly helpful if this worked. (in addition to not segfaulting. The segfaulting issue is much more basic, so I made a separate issue for that) I think this requires changing both Magma and CoreIR. See https://github.com/rdaly525/coreir/issues/684 for the matching CoreIR issue.

leonardt, 2019-02-13:
Support for working with internal signals has been added to fault  (peeking nested instance ports, poking nested registers), so I would recommend migrating to that interface for future tests. 

However this only works for the verilator/ncsim/vcs targets right now. From what I understand, the coreir simulator doesn't support accessing internal signals yet. Once this API is exposed, adding support in fault should be straightforward.

David-Durst, 2019-02-13:
Can I breakpoint mid test in fault?

leonardt, 2019-02-13:
Nope, sorry that's an outstanding issue, see https://github.com/leonardt/fault/issues/19

David-Durst, 2019-02-13:
I will not be switching over until I can breakpoint. The tests take multiple minutes to run. I need the ability to stop and look around in a debugger.