phanrahan/magma #373: Install error for magma 1.0 and 0.1.20

kentaroy47, 2019-03-29:
I got a install error trying magma 1.0 and magma 0.1.20.
However, the install goes well if I move to magma 0.1.18.

env python==3.6.1, Ubuntu 16.04

```
pip install -e .
Obtaining file:///home/ken/magma
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/ken/magma/setup.py", line 39, in <module>
        long_description_content_type="text/markdown"
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/site-packages/setuptools/__init__.py", line 137, in _install_setup_requires
        dist.parse_config_files(ignore_option_errors=True)
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/site-packages/setuptools/dist.py", line 702, in parse_config_files
        self._parse_config_files(filenames=filenames)
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/site-packages/setuptools/dist.py", line 599, in _parse_config_files
        (parser.read_file if six.PY3 else parser.readfp)(reader)
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/configparser.py", line 718, in read_file
        self._read(f, source)
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/configparser.py", line 1015, in _read
        for lineno, line in enumerate(fp, start=1):
      File "/home/ken/miniconda3/envs/magma/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 71: ordinal not in range(128)
   ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/ken/magma/
```

leonardt, 2019-03-29:
The latest magma release depends on Python 3.7.2, I believe the change were merged in before we bumped the major version number (1.0), which would explain why you're having issues with 0.1.20. Can you try upgrading your Python version to see if that resolves the issue?

kentaroy47, 2019-03-29:
@leonardt 
yes, by going to Python==3.7.2, the install went fine.
Thanks for the answers!

closed 2019-03-29 by kentaroy47