phanrahan/magma #388: Regression in wrapped verilog compile

leonardt, 2019-05-02:
Using latest magma/fault master

```
$ cd fault
$ pytest -k "test_tester_verilog_wrapped[verilator-None]"
============================= test session starts ==============================
platform darwin -- Python 3.7.2, pytest-4.4.0, py-1.8.0, pluggy-0.9.0
rootdir: /Users/lenny/repos/fault, inifile: setup.cfg
plugins: cov-2.6.1, codestyle-1.4.0
collected 68 items / 67 deselected / 1 selected

tests/test_tester.py F                                                   [100%]

=================================== FAILURES ===================================
_________________ test_tester_verilog_wrapped[verilator-None] __________________

target = 'verilator', simulator = None

    def test_tester_verilog_wrapped(target, simulator):
        SimpleALU = m.DefineFromVerilogFile("tests/simple_alu.v",
                                            type_map={"CLK": m.In(m.Clock)},
                                            target_modules=["SimpleALU"])[0]
    
        circ = m.DefineCircuit("top",
                               "a", m.In(m.Bits[16]),
                               "b", m.In(m.Bits[16]),
                               "c", m.Out(m.Bits[16]),
                               "config_data", m.In(m.Bits[2]),
                               "config_en", m.In(m.Bit),
                               "CLK", m.In(m.Clock))
        simple_alu = SimpleALU()
        m.wire(simple_alu.a, circ.a)
        m.wire(simple_alu.b, circ.b)
        m.wire(simple_alu.c, circ.c)
        m.wire(simple_alu.config_data, circ.config_data)
        m.wire(simple_alu.config_en, circ.config_en)
        m.wire(simple_alu.CLK, circ.CLK)
        m.EndDefine()
    
        tester = fault.Tester(circ, circ.CLK)
        tester.verilator_include("SimpleALU")
        tester.verilator_include("ConfigReg")
        for i in range(0, 4):
            tester.poke(
                fault.WrappedVerilogInternalPort("SimpleALU_inst0.config_reg.Q",
                                                 m.Bits[2]),
                i)
            tester.step(2)
            tester.expect(
                fault.WrappedVerilogInternalPort("SimpleALU_inst0.opcode",
                                                 m.Bits[2]),
                i)
            signal = tester.peek(
                fault.WrappedVerilogInternalPort("SimpleALU_inst0.opcode",
                                                 m.Bits[2]))
            tester.expect(
                fault.WrappedVerilogInternalPort("SimpleALU_inst0.opcode",
                                                 m.Bits[2]),
                signal)
            tester.expect(
                fault.WrappedVerilogInternalPort(
                    "SimpleALU_inst0.config_reg.Q", m.Bits[2]),
                i)
            signal = tester.peek(
                fault.WrappedVerilogInternalPort(
                    "SimpleALU_inst0.config_reg.Q", m.Bits[2]))
            tester.expect(
                fault.WrappedVerilogInternalPort(
                    "SimpleALU_inst0.config_reg.Q", m.Bits[2]),
                signal)
[... 129 lines cut ...]
m.compile("tmp", circ, output="coreir-verilog")
```

Result of running it in a file `tmp.py`
```
❯ python tmp.py
/Users/lenny/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collect
ions.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
/Users/lenny/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collect
ions.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
/Users/lenny/repos/coreir/src/binary/coreir.cpp:187 Running Runningvpasses
/Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/lenny/repos/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
  Def:
    Instances:
      ConfigReg_inst0 : ConfigReg
    Connections:

ERROR: {SimpleALU}.self Is not connected

I AM DYING!
Assertion failed: (0), function die, file /Users/lenny/repos/coreir/src/ir/context.cpp, line 112.
```

rsetaluri, 2019-05-02:
I think it just needs https://github.com/rdaly525/coreir/pull/735/ to go through

Kuree, 2019-05-06:
I've seen this bug in my garnet flow. It's resolved using pycoreir release version instead of the one compiled from scratch, see https://github.com/Kuree/GarnetFlow/blob/master/scripts/install.sh#L25-L26

rsetaluri, 2019-05-06:
Is this still an issue?

leonardt, 2019-05-06:
Fixed for me by latest pycoreir release

closed 2019-05-06 by leonardt