phanrahan/magma #393: Circuit That Used To Compile To Verilog No Longer Does

David-Durst, 2019-05-12:
Back in February, https://github.com/David-Durst/aetherling/blob/master/tests/haskell/test_downsampleStencil_big.py#L83 used to pass. Now, it fails with the error below.

The only changes I've made regarding the test are converting Bits() to Bits[] and Array() to Array[] in https://github.com/David-Durst/aetherling/blob/master/tests/haskell/downsample_256x256_to_32x32_2px_in_per_clk.py. You can see the diff on the file, it's tiny and has not impact.

My linebuffer works as it passes all my CoreIR simulator tests.

Regarding the specific error about a 1-D array of bits: 
1. My outer module's interface is flattened. Why does Magma need to care about the rest?
2. I'm running the CoreIR flatten passes, so the interfaces the verilog backend will see are 1-D

```
_______________ test_downsample_256x256_to_32x32_2px_in_per_clk ________________

    def test_downsample_256x256_to_32x32_2px_in_per_clk():
        from .downsample_256x256_to_32x32_2px_in_per_clk import c, downsample_256x256_to_32x32_2px_in_per_clk as testcircuit
    
        magma.compile("vBuild/" + testcircuit.name, testcircuit, output="verilog",
                      passes=["rungenerators", "wireclocks-coreir", "verifyconnectivity --noclkrst", "flattentypes", "flatten", "verifyconnectivity --noclkrst", "deletedeadinstances"],
>                     namespaces=["aetherlinglib", "commonlib", "mantle", "coreir", "global"], context = c)

test_downsampleStencil_big.py:88: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../magma/magma/compile.py:130: in compile
    write_file(file_name, suffix, verilog.compile(main))
../../../magma/magma/backend/verilog.py:253: in compile
    code += compiledefinition(v) + '\n'
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

cls = MapParallel_n8_opSIPO1CE_I_In_Bit__O_Out_Bits_1___CLK_In_Clock__CE_In_Enable__ = DefineCircuit("MapParallel_n8_opSIPO1..., Out(Bit)]])
wire(None, MapParallel_n8_opSIPO1CE_I_In_Bit__O_Out_Bits_1___CLK_In_Clock__CE_In_Enable__.O)
EndCircuit()

    def compiledefinition(cls):
    
        # for now only allow Bit or Array(n, Bit)
        for name, port in cls.interface.ports.items():
            if isinstance(port, (ArrayKind, ArrayType)):
                if not isinstance(port.T, (_BitType, _BitKind)):
>                   raise Exception(f'Argument {cls.__name__}.{name} of type {type(port)} is not supported, the verilog backend only supports simple 1-d array of bits of the form Array(N, Bit)')
E                   Exception: Argument MapParallel_n8_opSIPO1CE_I_In_Bit__O_Out_Bits_1___CLK_In_Clock__CE_In_Enable__.O of type Array[8, Array[1, In(Bit)]] is not supported, the verilog backend only supports simple 1-d array of bits of the form Array(N, Bit)

../../../magma/magma/backend/verilog.py:163: Exception
```

rdaly525, 2019-05-13:
@leonardt It looks like you check to make sure that magma circuits are only allowed to be compiled to verilog (via coreir) if they have ports of type Array(Bit) or Bit. Am I misunderstanding the code, or is this intended/NYI?

Circuits can still be compiled to verilog as long as the coreir pass "flattentypes" is called.

leonardt, 2019-05-13:
This test is failing inside 
```
../../../magma/magma/backend/verilog.py:253: in compile
    code += compiledefinition(v) + '\n'
```

which means you're using the magma verilog backend (as opposed to the coreir verilog backend). Is this intended? I suspect not, because you're passing arguments such as `passes=["rungenerators", "wireclocks-coreir", "verifyconnectivity --noclkrst", "flattentypes", "flatten", "verifyconnectivity --noclkrst", "deletedeadinstances"]` to the `m.compile` command.

If you're intending to use the coreir backend, can you try setting `output="coreir-verilog"` instead of `"verilog"`? I'm not sure why the behavior of `compile` changed, but we did add some logic to use `coreir-verilog` automatically if `mantle_target == "coreir"` so perhaps something related to that code path caused this regression.

David-Durst, 2019-05-13:
Success! Changing from `output="verilog"` to `output="coreir-verilog"` made the test pass (after a few changes to my fault interface).

closed 2019-05-13 by David-Durst