phanrahan/magma #407: Reusing CoreIR Instance Across Unit Tests Gives Wrong CoreIR Module Signature

David-Durst, 2019-06-11:
This issue is based on the magma branch https://github.com/phanrahan/magma/tree/cache_coreir_context and mantle branch https://github.com/phanrahan/mantle/tree/cache_coreir_context where I use a singleton CoreIR context and CoreIR backend.

The singletons are causing issues.

When I individually run each of the tests in https://github.com/David-Durst/aetherling/blob/master/tests/test_delayed_buffer.py, they pass.

However, when I run them both at once with the command `pytest -s tests/test_delayed_buffer.py`, they fail. The error log is below.

The issue is that the singleton CoreIRBackend and CoreIR context are returning the incorrect module when [import generator is called](https://github.com/phanrahan/magma/blob/cache_coreir_context/magma/frontend/coreir_.py#L32). I've added code at https://github.com/phanrahan/magma/blob/cache_coreir_context/magma/frontend/coreir_.py#L34-L38 to compare what the singletons return compared to a fresh CoreIRBackend and CoreIR context.

The line in the below error log shows the difference. 'in' is the correct name, not 'in_0'. For the CoreIR module that is being wrapped, it is impossible for any of the ports to have the name 'in_0'.
```
Cached cirb IO: dict_items([('in_0', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10cb3f390>)])
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfd9630>), ('out', <coreir.type.Type object at 0x10d1ab2b0>)])
F
```

The error log is
```
Testing started at 13:05 ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/183.4886.43/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --path /Users/durst/dev/W17-8/aetherling/tests/test_delayed_buffer.py -- -s
Launching pytest with arguments -s /Users/durst/dev/W17-8/aetherling/tests/test_delayed_buffer.py in /Users/durst/dev/W17-8/aetherling/tests

/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
============================= test session starts ==============================
platform darwin -- Python 3.7.1, pytest-4.4.1, py-1.8.0, pluggy-0.9.0
rootdir: /Users/durst/dev/W17-8/aetherling
plugins: xdist-1.28.0, forked-1.0.2collected 2 items

test_delayed_buffer.py Cached cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10cb3f390>)])
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10cb3fba8>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Cached cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfadfd0>), ('out', <coreir.type.Type object at 0x10cfad080>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfadfd0>), ('out', <coreir.type.Type object at 0x10cfb2128>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Cached cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfd9630>), ('out', <coreir.type.Type object at 0x10cb3f390>)])
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfd9630>), ('out', <coreir.type.Type object at 0x10cfd9a20>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Found raddr
Starting topological sort
topo_order.size() = 64
numVertices(g)    = 64
.Cached cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10cb3f390>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10d0d37f0>)])
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Cached cirb IO: dict_items([('in_0', <coreir.type.Type object at 0x10cb3f7b8>), ('out', <coreir.type.Type object at 0x10cb3f390>)])
Fresh cirb IO: dict_items([('in', <coreir.type.Type object at 0x10cfd9630>), ('out', <coreir.type.Type object at 0x10d1ab2b0>)])
F
tests/test_delayed_buffer.py:40 (test_delayed_buffer_parallel)
[... 59 lines cut ...]
cls = Mux_Array_1_Array_8_Bit__t_2n = DefineCircuit("Mux_Array_1_Array_8_Bit__t_2n", "data", Array[2, Array[1, Array[8, In(B...t___in_0_Array_8_In_Bit___out_Array_8_Out_Bit____inst0.in_0)
wire(None, Mux_Array_1_Array_8_Bit__t_2n.out)
EndCircuit()

    @classmethod
    def definition(cls):
        if n > 1:
            type_size_in_bits = GetCoreIRBackend().get_type(t).size
            mux = CommonlibMuxN(n, type_size_in_bits)
    
            type_to_bits = DefineNativeMapParallel(n, DefineDehydrate(t))()
>           wire(cls.data, type_to_bits.I)
E           AttributeError: 'NativeMapParallel_n2_opdehydrate_tArray_1_Array_8_' object has no attribute 'I'

../aetherling/modules/mux_any_type.py:35: AttributeError
=============================== warnings summary ===============================
tests/test_delayed_buffer.py::test_delayed_buffer_serial
tests/test_delayed_buffer.py::test_delayed_buffer_parallel
  /Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/hwtypes/bit_vector_abc.py:45: UserWarning: DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecated
    warnings.warn('DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecated')

-- Docs: https://docs.pytest.org/en/latest/warnings.html
================ 1 failed, 1 passed, 2 warnings in 3.13 seconds ================
Process finished with exit code 0
```

rdaly525, 2019-06-11:
This is caused by the pass 'flattentypes' being called and then recalling the type generator function which will produce different results. 'flattentypes' causes the typegen associated with a generator to be invalid. There should be a better way to handle this in coreir.

David-Durst, 2019-06-11:
Calling [__reset_context](https://github.com/phanrahan/magma/blob/cache_coreir_context/magma/backend/coreir_.py#L74) in [conftest.py](https://github.com/David-Durst/aetherling/blob/master/conftest.py) fixed the issue of getting the wrong module back.

However, __reset_context caused another issue. Now, both the unit tests fail when run individually with the error `E           Exception: Got None for port 'dehydrate_tArray_8_Bit_.out', is it connected to anything?`. 

```
Testing started at 13:51 ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/183.4886.43/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --target test_delayed_buffer.py::test_delayed_buffer_serial -- -s
Launching pytest with arguments -s test_delayed_buffer.py::test_delayed_buffer_serial in /Users/durst/dev/W17-8/aetherling/tests

/Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
============================= test session starts ==============================
platform darwin -- Python 3.7.1, pytest-4.4.1, py-1.8.0, pluggy-0.9.0
rootdir: /Users/durst/dev/W17-8/aetherling
plugins: xdist-1.28.0, forked-1.0.2collected 1 item

test_delayed_buffer.py /Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/durst/dev/W17-8/coreir/2coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
F
tests/test_delayed_buffer.py:11 (test_delayed_buffer_serial)
def test_delayed_buffer_serial():
        scope = Scope()
        testcircuit = DefineDelayedBuffer(Array[8, Bit], 3, 1, 12)
    
>       sim = CoreIRSimulator(testcircuit, testcircuit.CLK)

test_delayed_buffer.py:16: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../magma/magma/simulator/coreir_simulator.py:141: in __init__
    coreir_.compile(circuit, coreir_filename, context=self.ctx)
../../magma/magma/backend/coreir_.py:461: in compile
    backend.compile(main)
../../magma/magma/backend/coreir_.py:435: in compile
    self.compile_dependencies(defn_or_declaration)
../../magma/magma/backend/coreir_.py:424: in compile_dependencies
    self.modules[key.name] = self.compile_definition(key)
../../magma/magma/backend/coreir_.py:320: in compile_definition
    self.compile_definition_to_module_definition(definition, module_definition)
../../magma/magma/backend/coreir_.py:280: in compile_definition_to_module_definition
    self.connect_non_outputs(module_definition, port, non_input_ports)
../../magma/magma/backend/coreir_.py:292: in connect_non_outputs
    non_input_ports)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <magma.backend.coreir_.CoreIRBackend object at 0x110212080>
module_definition = <coreir.module.ModuleDef object at 0x1109a7b70>
port = dehydrate_tArray_8_Bit_.out, value = None
non_input_ports = {dehydrate_tArray_8_Bit_.in: 'self.in', dehydrate_tArray_8_Bit_.in[0]: 'self.in.0', dehydrate_tArray_8_Bit_.in[1]: 'self.in.1', dehydrate_tArray_8_Bit_.in[2]: 'self.in.2', ...}

    def connect(self, module_definition, port, value, non_input_ports):
        self.__unique_concat_id
    
        # allow clocks or arrays of clocks to be unwired as CoreIR can wire them up
        def is_clock_or_nested_clock(p):
            if isinstance(p, (ClockType, ClockKind)):
                return True
            elif isinstance(p, (ArrayType, ArrayKind)):
                return is_clock_or_nested_clock(p.T)
[... 59 lines cut ...]
                for item in p.Ts:
                    if is_clock_or_nested_clock(item):
                        return True
            return False
    
        if value is None and is_clock_or_nested_clock(port):
            return
        elif value is None:
            if port.isinout():
                # Skip inouts because they might be connected as an input
                return
>           raise Exception(f"Got None for port '{port.debug_name}', is it "
[... cut ...]
=============================== warnings summary ===============================
tests/test_delayed_buffer.py::test_delayed_buffer_serial
  /Users/durst/anaconda/envs/aetherling/lib/python3.7/site-packages/hwtypes/bit_vector_abc.py:45: UserWarning: DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecated
    warnings.warn('DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecated')

-- Docs: https://docs.pytest.org/en/latest/warnings.html
===================== 1 failed, 1 warnings in 0.57 seconds =====================
Process finished with exit code 0

```

David-Durst, 2019-06-11:
I fixed the issue with https://github.com/phanrahan/magma/commit/c939e06c864956a837622d46249c9cb7fc4edb93.

The global variable pointing to a CoreIR context wasn't being reset properly in conftest. Some old references to the variable were being maintained. I replaced it with a singleton class.

closed 2019-06-11 by David-Durst