phanrahan/magma #414: Error compiling top level DefineFromVerilog

leonardt, 2019-06-17:
Verilog source example
```verilog
// counter.v
module vcounter(input CLK, output [2:0] O);
reg [2:0] O;
always @(posedge CLK) begin
    O <= O + 1'b1;
end
endmodule
```

Magma top example
```python
# top.py
import magma as m

circ = m.DefineFromVerilogFile("counter.v", target_modules=["vcounter"])[0]
m.compile("counter2", circ, output="coreir-verilog")
```

Errors out with coreir module has no def
```bash
❯ python top.py
ERROR: Module: global.vcounter
  Type: {'CLK':BitIn, 'O':Bit[3]}
  Def? No has no def!

0   libcoreir.dylib                     0x000000010bdf0d48 _ZN6CoreIR7Context6setTopEPNS_6ModuleE + 104
1   libcoreir.dylib                     0x000000010bef5af6 _ZN6CoreIR12loadFromFileEPNS_7ContextENSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9al
locatorIcEEEEPPNS_6ModuleE + 86806
2   coreir                              0x000000010bc5238d main + 24077
3   libdyld.dylib                       0x00007fff7699d3d5 start + 1
```

@rsetaluri do you know if this is a magma related issue or a coreir issue? My hunch is that coreir needs to recognize when the top level definition as a "wrapped verilog" module.

I see that magma produces the .json correctly
```json
{"top":"global.vcounter",
"namespaces":{
  "global":{
    "modules":{
      "vcounter":{
        "type":["Record",[
          ["CLK","BitIn"],
          ["O",["Array",3,"Bit"]]
        ]],
        "metadata":{"verilog":{"verilog_string":"module vcounter(input CLK, output [2:0] O);\nreg [2:0] O;\nalways @(posedge CLK) begin\n    O <= O + 1'b1;\nend\nendmodule\n"}}
      }
    }
  }
}
}
```

So this suggests the error is occurring during the invocation of `coreir -i <file>.json -o <file>.v` which would suggest that it can't handle the above json file (which seems okay assuming we're allowing top-level black boxes).

This isn't high priority as the workaround is just use magma's `verilog` backend to emit the wrapped verilog, but it would be nice to have this end to end integration working.

rdaly525, 2019-06-17:
I suspect the issue is that I check to make sure the top module has a definition. But CoreIR does not (currently) recognize that the verilog metadata is a definition.

rdaly525, 2019-06-18:
A hacky way to fix this is to add an empty coreir definition to the module.

rsetaluri, 2019-06-19:
Yes, the issue is that CoreIR always requires the top level module to have a definition, and verilog metadata is not considered "a definition". If you look at the code CoreIR defines having a definition as having a non-empty `connections` field in the json.